supermerill/SuperSlicer #1159: Temperature offset setting does not affect temperature calibration

JPTa, 2021-05-20:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
I have set extruder temp offset to -15 but it is not being used when generating a temperature calibration tower via the Calibration menu -> Filament temperature calibration. To get proper values in the gcode I have to manually edit each one of them.

supermerill, 2021-05-20:
can you post your project file ( just before clicking on export gcode) ?

JPTa, 2021-05-20:
Here's the project file. 
[smart_compact_temp_calibration.zip](https://github.com/supermerill/SuperSlicer/files/6518056/smart_compact_temp_calibration.zip)

In generated gcode the first wrong setting is on line 375: M104 S230 ; ground floor temp tower set
That should be S215 as the extruder offset is -15 being the extruder_temperature_offset[0].

supermerill, 2021-05-20:
can you past a screen shot of 
[image]
?

For what i see, the number in gcode is the one that is written in the tower.

JPTa, 2021-05-20:
Down 3  up 2 steps 5

JPTa, 2021-05-20:
The extruder_temperature_offset setting does affect the temperatures elsewhere but not in the generated settings for the temperature tower. The default temp for this filament in the filament profile is 220 and the first layer is 225. In reality this printer's extruder temperature sensor outputs +15C too high temperatures hence the use of the offset for the printer profile. The floor temp in this test was 230 and that should read in the printed tower but in reality (gcode) it should be 215 if the extruder offset was used as it should be when the tower is generated.

JPTa, 2021-05-20:
The problem propably is in file  SuperSlicer/src/slic3r/GUI/CalibrationTempDialog.cpp starting around line 69, where the current temperature is fetched from the filament profile but the offset is ignored and not added to temperature.

supermerill, 2021-05-21:
I understand.
It's a bit hard to do, as the temperature calibration use some custom code that currently aren't parsed. So the offset isn't applied on them