supermerill/SuperSlicer #1160: Loss of most recent presets

bakaufman, 2021-05-21:
**Describe the bug**
My Printer Settings added since 2.3.55.3 are not available when I launch compile #260 for mac.

**To Reproduce**
Steps to reproduce the behavior:
1. Have had SS for a while, before and after the advent of Physical printer. Have used a version with physical printer
2. Install #260 separately.
3. open new file or previously saved .3mf files
4. Any Printer settings, Filament Settings, Printer Settings added in 2.3.55.3 are not available

**>> Project File <<**
Doesn't seem necessary

**Expected behavior**
I expected this update/upgrade to find and use prior presets

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Version [e.g. 2.3.55.5], 2.3.56.0 #260

**Additional context**
Add any other context about the problem here.

supermerill, 2021-05-21:
This version switches back from the `superslicer-alpha` to `superslicer` setting directory, maybe that's that?

bakaufman, 2021-05-21:
that is what it looks like and what I speculated in the discord. Can we have options around this?

bakaufman, 2021-05-21:
might convert this to a feature request, or at least find a way to synchronize these folders if we are going to flip back and forth.

supermerill, 2021-05-21:
you can just rename it.
i'll stop changing it.

bakaufman, 2021-05-21:
That worked and is nice. I'll close it.

closed 2021-05-21 by bakaufman