supermerill/SuperSlicer #1161: [2.3.65.0] UI encoding issue like "lÃ¶st", at least with german language setting

dartrax, 2021-05-22:
**Describe the bug**
Looks like an encoding issue in this version, in 2.3.55 everything was fine. Now characters like ä ö ü ß are shown as Ã¶ all over the interface. This also affects search function (not possible to search settings with special characters).

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win 10 64bit
 - Version [2.3.65.0]

supermerill, 2021-05-22:
can't replicate
[image]

maybe because i set my windows into beta-utf-8 mode.
I'll see if I can reproduce it on an other computyer

dartrax, 2021-05-22:
It must have something to do with the files in \resources\localization\de\*.

When I replace the files with the ones of Version 2.2.55.5, everything is ok.

supermerill, 2021-05-22:
I changed where they are done (not on my computer anymore).
Maybe it's the problem.
I wonder if the same problem will arise in other os...

supermerill, 2021-05-22:
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/6526455/SuperSlicer.zip)
 can you test with that one?

dartrax, 2021-05-22:
How to create the .mo file, is it simply saving the Slic3r.po with Poedit and renaming the generated Slic3r.po to SuperSlicer.mo? I've tried that and it worked.

Edit: Ah, you are fast. Yes, that's working now! No more encoding issue 👍

jfestrada, 2021-05-22:
Hi, I can confirm that there is the same issue with the spanish translation on a win64. The linux version is working fine, so is only related with the windows build.

Windows version:

[image]

Linux version (PopOS amd64):

[image]

dartrax, 2021-05-22:
Try this one:
[SuperSlicer_ES.zip](https://github.com/supermerill/SuperSlicer/files/6527230/SuperSlicer_ES.zip)

jfestrada, 2021-05-22:
> 
> 
> Try this one:
> [SuperSlicer_ES.zip](https://github.com/supermerill/SuperSlicer/files/6527230/SuperSlicer_ES.zip)

This file solve the issue for the spanish translation.

[image]

supermerill, 2021-05-22:
good to know.. I'll have to manually create the translation for windows.

closed 2021-06-13 by supermerill