supermerill/SuperSlicer #1163: 2.3.56 - Very low frame rate

JasonGuthrie, 2021-05-22:
Any project (no matter how simple) is suffering from very low frame rates, especially once sliced. 

Using RivaTuner to track the FPS, I noted a max of 12fps with a simple sliced object (and SS became effectively unresponsive with a larger and more complex model). 
The same projects happily runs at 60 fps (my monitors max refresh rate) in other PS/SS versions. 

I am using a GTX 1060 with the most recent studio driver (462.59), which is experiencing little to no load despite the low FPS.

I can't further test the latest release of SS unfortunately due this issue.

supermerill, 2021-05-22:
your os?

JasonGuthrie, 2021-05-22:
W10 Pro - 10.0.19042 Build 19042

I didn't try a cube, just a project file I was working on prior but I have the same performance issue with default box (in fact its worse with the cube @ 4 FPS).

Also, as a related side note, the resolution on the default build plate graphic is very low. I can't quantify it but it is now the equivalent of 480p where the prevision version was 1080p (if you know what I mean).

CCS86, 2021-05-23:
I'm pretty sure this is just a byproduct of a different bug I just reported: https://github.com/supermerill/SuperSlicer/issues/1164

wildjokerss, 2021-05-23:
and I thought my pc was acting up :D
Yep. I'm also having this issue. It's almost unusable to me. (i7 6700k + GTX 1070 + 32gb RAM) I'd imagine people with lower specs are still processing the 3rd frame by now :D

MLiv79, 2021-05-23:
Yes also for me is completely unusable. It doesnt use the GPU, only CPU

MLiv79, 2021-05-23:
> your os?
> it happens even with no object loaded?
> `a simple sliced object` = a cube?

Windows, the problem as i wrote before is that it use only the CPU and not the GPU (0% load)

MLiv79, 2021-05-23:
This nightly of 3 days ago  work as expected
[image]

CCS86, 2021-05-23:
> I'm pretty sure this is just a byproduct of a different bug I just reported: #1164

I have changed my mind. The frame rate is definitely worse as a result of the bug I linked to. But, even in the normal 3D preview there is a frame rate and quality issue.

In 2.3.55.5, the GPU is being used and AA is on. In 2.3.56.0, the CPU is drawing and no AA is present.

supermerill, 2021-05-24:
there is a `opengl32.dll` that shouldn't be here (in the root directory). Delete it.

JasonGuthrie, 2021-05-24:
> there is a `opengl32.dll` that shouldn't be here (in the root directory). Delete it.

Aye that's fixed both the blurry build plate and low frame rates. Keep up the good work.

closed 2021-05-24 by JasonGuthrie