supermerill/SuperSlicer #1164: 2.3.56: Extremely Small Print Segments

CCS86, 2021-05-23:
SS 2.3.56 / Win 10 x64

When compared to 2.3.55, with identical settings, 2.3.56 is creating around 6 times the number of print segments.

I believe this is also responsible for poor frame rate when in the sliced preview.

Load the attached project in both versions, and compare the output gcode, or displayed number of print segments.

[Right Duct.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6527337/Right.Duct.3mf.txt)

I realize that I have modified settings like 'minimum extrusion length', but reverting this did not improve the issue. I have tried a few others, like 'resolution' and 'model rounding precision', but haven't found one that solves it.

macchinette, 2021-05-23:
Hi, I had the same problem, the resolution accuracy has been increased and where before you had 0 you have to put a value that your printer can handle.

[image]

CCS86, 2021-05-23:
Good find. I didn't notice that at first, because gap filling seems to ignore it.

When I turn off gap filling, setting 2.3.56.0 to resolution=0.013mm seems to give an equivalent number of print segments to 2.3.55.5 with resolution=0

 Interestingly, it seems to be wasting segments somewhere, because in an identical section of the print 2.3.56 is only using 8 segments to form a curve, where 2.3.55 is using 10:

[image]

[image]

When you make no other change, but enable gap filling, this is the change to total print segments:

2.3.55:   13226 >>>>> 19783    An increase of 50%

2.3.56:   13156 >>>>>  33243  An increase of 153%

So, it seems that the 'resolution' setting is more of a bandaid  to something wrong in the engine, than it is a solution.

CCS86, 2021-05-23:
With resolution set to 0 (like it has been), the gap filling segments get really bad:

https://user-images.githubusercontent.com/12439543/119273563-12a08200-bbd1-11eb-89ff-2678588956f2.mp4

supermerill, 2021-05-24:
I think that's because previously, the `printer-> min extrusion length` wasn't working quite reliably.
Now it should be better.
With `print -> resolution : 0`, the smallest segment will be of the size of the smallest chunk of triangle from the stl.
So the only thing that will "reduce" the final resolution is the `printer-> min extrusion length`, which is currently at 0.006mm
If you change it to something more reasonable like 0.05, you'll see something better.

Unless i forgot something.

CCS86, 2021-05-25:
Ah, I forgot that I came up with that 0.006mm min extrusion length as a compensation for it not working as described.

Bumping that min extrusion length way up to 0.1mm, and leaving the resolution set to 0 is giving a much better result!

closed 2021-05-25 by CCS86