supermerill/SuperSlicer #1165: SuperSlicer_2.2.56.0_win64_20210522.zip for release  2.3.56.0 ? Why the discrepancy in versioning?

xgpt, 2021-05-23:
**Describe the bug**

Possibly wrong versioning on latest pre-release.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://github.com/supermerill/SuperSlicer/releases/tag/2.3.56.0'
2. Click on 'assets'
3. Scroll down to 'the windows one'
4. See error:

```
SuperSlicer-ubuntu_18.04-2.2.56.0.AppImage
70 MB
SuperSlicer_2.2.56.0_macos_20210522.dmg
46.1 MB
SuperSlicer_2.2.56.0_win64_20210522.zip
55.2 MB
SuperSlicer_2.3.56.0_linux64_20210522.tar.zip
68.2 MB
Source code (zip)
Source code (tar.gz) 
```

**Expected behavior**
Please update windows binary zip to correct version number on Git

**Screenshots**
It's correctly 2.3.56 in the actual Windows binary.

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 2.3.55.5], Use `Help->About`_
2.3.56.0

**Additional context**
I think this happened before too, very confusing. Please consider making a release checklist and adding "update version number? in this place, that place, and that other place" to it. I personally thrive on little checklists like that to make sure I don't make little silly mistakes like this! Love the release though! Still a little confused about spacing vs. widths but I LOVEEEE That there's a new help/info thing in the actual app with an explanation of function! I'D LOVEEEE to see more "help" tips on sooo much more. The way it's laid out I think it's the next way to make the best slicer also the most user accessible/friendly to all levels of 3d tinkerers! <3 it! Please ping me if you need help coming up with documentation drafts!

ldiegos, 2021-05-23:
Not only on last, same issue with SuperSlicer_2.3.55.0_win64_20210522.zip the folder inside is the SuperSlicer_2.2.53.0_win64_20210522 and for me this produce a lot of mistakes when trying to update your version.

epson950, 2021-05-23:
Поддерживаю. Просим название файла сохранять от названия версии, а не на порядок старше.

CCS86, 2021-05-23:
It's just a typo in the zip name.

epson950, 2021-05-23:
> 
> 
> It's just a typo in the zip name.

Already several versions? You have to be more careful, users get from this problem.

supermerill, 2021-05-24:
><

supermerill, 2021-05-27:
ok, I've made the release script (in the root directory).
Now, it should be consistent because I have nothing to type, and so nothing to fail.
If there are still some errors in the latest release, re-open this issue and I'll fix the script.

closed 2021-05-27 by supermerill