supermerill/SuperSlicer #1166: A debugging check in this application failed 2.3.56

Jack77777777, 2021-05-23:
**Describe the bug**
See error in screenshots section

**To Reproduce**
Steps to reproduce the behavior:
1. Open SuperSlicer 2.3.56

**>> Project File <<**
[Project.zip](https://github.com/supermerill/SuperSlicer/files/6527497/Project.zip)

**Expected behavior**
No error

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro; 20H2; 19042.985; Windows Feature Experience Pack 120.2212.2020.0
 - Version 2.3.56

tome9111991, 2021-05-23:
same probleme here.
and btw on german version problems with ö,ä,ü

[image]

Jack77777777, 2021-05-23:
> Program crash
> i try change temperature first layer extruder or
> Bed other layers
> Only when i type a value with keyboard.
> If i use arrow got debugging check error

Can confirm!

cbc02009, 2021-05-23:
Same issue here.

supermerill, 2021-05-23:
can someone test the build https://github.com/supermerill/SuperSlicer/actions/runs/869422325 ?
it's a debug build, so it's a bit slower.

cbc02009, 2021-05-23:
> 
> 
> can someone test the build https://github.com/supermerill/SuperSlicer/actions/runs/869422325 ?
> it's a debug build, so it's a bit slower.

Works for me!

tome9111991, 2021-05-24:
> can someone test the build https://github.com/supermerill/SuperSlicer/actions/runs/869422325 ?
> it's a debug build, so it's a bit slower.

works for me too
no error no crash

Liger0, 2021-05-24:
> 
> 
> can someone test the build https://github.com/supermerill/SuperSlicer/actions/runs/869422325 ?
> it's a debug build, so it's a bit slower.
Crash at start
[image]

bttguy, 2021-05-25:
Can we expect a prebuilt zip hotfix release for it, or we should wait for the next release you planned?

Jack77777777, 2021-05-26:
> Can we expect a prebuilt zip hotfix release for it, or we should wait for the next release you planned?

The latest version is working for me when Unicode UTF-8 is enabled in Windows. My Windows is set to german and SS to english.

Control Panel->Clock and Region->Region->Administrative TAB->Change system locale...->Check **Beta: Use Unicode UTF-8 for worldwide language support**

supermerill, 2021-05-26:
> Can we expect a prebuilt zip hotfix release for it, or we should wait for the next release you planned?

you can use the nightly. The windows build just miss the poedit-compiled localization files.

Liger0, 2021-05-26:
> 
> I'll do a release this evening if no other bug is reported.

uhm, what about the crash I reported?

supermerill, 2021-05-26:
> > I'll do a release this evening if no other bug is reported.
> 
> uhm, what about the crash I reported?

Which one?
If it's about the italian translation, I can't reproduce your error, as i don't have the file.
The merging tool is repaired (in the nightly).
If you have a new file, i can include it. Do I include the one in https://github.com/supermerill/SuperSlicer/issues/1128#issuecomment-846606648 ?

Liger0, 2021-05-26:
> 
> 
> > > I'll do a release this evening if no other bug is reported.
> > 
> > 
> > uhm, what about the crash I reported?
> 
> Which one?
> If it's about the italian translation, I can't reproduce your error, as i don't have the file.
> The merging tool is repaired (in the nightly).
> If you have a new file, i can include it. Do I include the one in [#1128 (comment)](https://github.com/supermerill/SuperSlicer/issues/1128#issuecomment-846606648) ?

Yes, the last one. If you are able to merge the TODO tell me, because the script tells me error and it fails to.
Also the crash is with any traslation, even old ones or the one included in the beta itself.

Liger0, 2021-05-27:
@supermerill I tried the new release "better spacing". Still same issue! 
[image]

bttguy, 2021-05-29:
Downloaded the latest build, have no issue since! Thanks the hotfix!

closed 2021-06-04 by supermerill