supermerill/SuperSlicer #1167: SuperSlicer_2.2.56.0_win64 italian language make program crash

chicco83, 2021-05-23:
if the software starts in italian it crash immediately.

MLiv79, 2021-05-23:
Yes and moreover you can't make it start in english. I've the previous version in english (because of known issues with italian language like fan settings) and the latest nightly force start in italian while i've not set it up in italian and like you wrote it force close immediately.

MLiv79, 2021-05-23:
Yes, i tried also latest release and result is the same. It start in italian and crash
[image]

chicco83, 2021-05-23:
to fix it simply copy eng file into it directory
here
SuperSlicer_2.2.56.0_win64_20210522\resources\localization

boromyr, 2021-05-23:
Try: https://github.com/supermerill/SuperSlicer/files/6523933/it.zip or https://github.com/supermerill/SuperSlicer/files/6433226/SuperSlicer.it.6.may.2021.zip

Liger0, 2021-05-23:
Use this nightly: https://github.com/supermerill/SuperSlicer/actions/runs/804896051
And the last posted files at #1128

supermerill, 2021-05-24:
It's on windows?
I tried each tabs on the italian just to be sure it won't crash.
And I can't make it crash.... maybe because my windows is in utf-8 mode... same problem as with other translations.
does this file works?
[it_2021_05_24.zip](https://github.com/supermerill/SuperSlicer/files/6534795/it_2021_05_24.zip)

Liger0, 2021-05-24:
> It's on windows?
> I tried each tabs on the italian just to be sure it won't crash.
> And I can't make it crash.... maybe because my windows is in utf-8 mode... same problem as with other translations. [...]

It still doesn't work.
Could you use as a base the updated one I posted yesterday in the last post, to see if anything need fixes? 
It works in the old nightly.
Plus there is translated stuff in the TODO but the python scripts fails so I can't merge it.

[image]

supermerill, 2021-05-27:
@Liger0 should be resolved, can you test the action debug_win#9?

supermerill, 2021-05-27:
nevermind

closed 2021-05-27 by supermerill