supermerill/SuperSlicer #1169: [2.3.56.0]Gcode preview window move part of a layer

ldiegos, 2021-05-23:
**Describe the bug**
The version 2.3.56.0 is moving part of a layer to a random place and this is written in the gcode. At least on my geeetech A10M configuration, dual color printer.

**To Reproduce**
Steps to reproduce the behavior:
1. Install version 2.3.56.0
2. Add a doubled stl for a dual color printer.
3. See Gcode preview
4. See error

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
Not to move the layers, and maintain the stl aspect.

**Screenshots**
[image]

Where it should be.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 pro
 - Version: 2.3.56.0

**Additional context**
Add any other context about the problem here.

supermerill, 2021-05-24:
Can't reproduce
[image]

Can i have your project file?

ldiegos, 2021-05-25:
Hi supermerill

Here are two project files.

The shift layer is on top of the model:
[dualTest3b.zip](https://github.com/supermerill/SuperSlicer/files/6540324/dualTest3b.zip)

The shift layer is in the back of the model:
[CaliCatDual.zip](https://github.com/supermerill/SuperSlicer/files/6540335/CaliCatDual.zip)

supermerill, 2021-05-25:
what's going on:
it see your `G1 Z3` and so it (wrongly) erase the next `G1 Z3` because the cleaner see this line as a duplicate.

I won't fix it as this part will go into a full rewrite, so may be fixed after the rewrite.

For now, you can change your `G1 Z3` by `G1 Z3.001`,

ldiegos, 2021-05-26:
Good morning supermerill!

Really the ToolChange G-code script made that? Let me try again.

Thank you so much.

ldiegos, 2021-05-26:
Perfect!
[image]

Perfect!
[image]

will you close the ticket? i will have in mind not to use common layer values and add always a .001 to the amount.

closed 2021-08-22 by supermerill