supermerill/SuperSlicer #1170: [2.3.56.0] UI encoding issue, at least with russian language setting

epson950, 2021-05-23:
[image]

ShohninDmitriy, 2021-05-23:
файлы киньте в в папку \resources\localization\ru\  предварительно удалив все из нее

[ru.zip](https://github.com/supermerill/SuperSlicer/files/6528792/ru.zip)

Alexdad76, 2021-05-23:
> 
> 
> файлы киньте в в папку \resources\localization\ru\ предварительно удалив все из нее
> 
> [ru.zip](https://github.com/supermerill/SuperSlicer/files/6528792/ru.zip)

Каракули исправляет. Надо бы чтобы по умолчанию English включался.
Doodle corrects. It would be necessary for the default English turned on.

epson950, 2021-05-23:
> 
> 
> файлы киньте в в папку \resources\localization\ru\ предварительно удалив все из нее
> 
> [ru.zip](https://github.com/supermerill/SuperSlicer/files/6528792/ru.zip)

Помогло! Спасибо! А продолжение над переводом будет? А то частично осталось на английском...

ShohninDmitriy, 2021-05-23:
я не участвовал в переводе просто прогнал ее через Poedit  чтоб каракули поправить

ShohninDmitriy, 2021-05-23:
[ru.zip](https://github.com/supermerill/SuperSlicer/files/6528965/ru.zip)

Частично обновил файлы оригиналов с Английского исходника

supermerill, 2021-05-24:
does this one works?
[ru_20210524.zip](https://github.com/supermerill/SuperSlicer/files/6534801/ru_20210524.zip)

surely the same problem as with others.

shadalik, 2021-05-24:
> does this one works?

Still does not work

Alexdad76, 2021-05-25:
> 
> 
> does this one works?
> [ru_20210524.zip](https://github.com/supermerill/SuperSlicer/files/6534801/ru_20210524.zip)
> 
> surely the same problem as with others.

I confirm - `ru_20210524.zip` still not working 
(Windows 7 64 bit)

supermerill, 2021-05-25:
I confirmed that gettext isn't creating a nice file for windows.
@ShohninDmitriy how did you transformed your .po in a .mo?

ShohninDmitriy, 2021-05-25:
> I confirmed that gettext isn't creating a nice file for windows.
> @ShohninDmitriy how did you transformed your .po in a .mo?

using the PoEdit program

supermerill, 2021-05-26:
Ok, the problem is the lack of header at the top of the .po to say "it's utf-8"
Will be fixed.

closed 2021-06-04 by supermerill