supermerill/SuperSlicer #1171: [2.3.55/2.3.56] DualColor/MixingNozzle: Missing a extrusion command the first time that the color/extruder is change

ldiegos, 2021-05-23:
**Describe the bug**
With my Geeetech A10M printer, with a mixing hotend, when i try to print a dual color stl(two stl) that have one color over the other, the first time that the idle extruder(the other) is called, superslicer miss to add a extrusion command and the first layer of the second color become weak in the union and break.

In the successive color change everything goes fine and the print is strong.

The first time that the idle extruder is called, the lines in the g-code are:

```
;-----Tool change g-code end-----------
; custom gcode end: toolchange_gcode
M106 S255 ; enable fan
T0 ; change extruder
G92 E0 ; reset extrusion distance
; custom gcode: start_filament_gcode
M900 K200; Filament gcode
; custom gcode end: start_filament_gcode
G1 X116.462 Y118.538 F7200.000 ; move to first Internal infill point
G1 Z3.200000 ; restore layer Z
                                                        ;<<<<<<<<<<<<<<<<<<<<<<<<<This line is missing, see the following gcode lines.
; custom gcode: feature_gcode
;-----Between extrusion role change g-code start-----------
;-----Between extrusion role change g-code end-----------
; custom gcode end: feature_gcode

;TYPE:Internal infill
G1 F2400.000
G1 X118.530 Y116.470 E0.09899 ; Internal infill
G1 X117.425 Y119.046 F7200.000 ; move to first Perimeter point
; custom gcode: feature_gcode
;-----Between extrusion role change g-code start-----------
;-----Between extrusion role change g-code end-----------
; custom gcode end: feature_gcode

;TYPE:Perimeter
```

The next time that the same extruder is called then:
```
;-----Tool change g-code end-----------
; custom gcode end: toolchange_gcode
M106 S255 ; enable fan
T0 ; change extruder
G92 E0 ; reset extrusion distance
; custom gcode: start_filament_gcode
M900 K200; Filament gcode
; custom gcode end: start_filament_gcode
G1 X116.462 Y118.538 F7200.000 ; move to first Internal infill point
G1 Z9.200000 ; restore layer Z
G1 E10.00000 F2400.000 ; unretract    ;<<<<<<<<<<<<<<<<<<<<<<<<<This line appear and everything goes perfect.
; custom gcode: feature_gcode
;-----Between extrusion role change g-code start-----------
;-----Between extrusion role change g-code end-----------
; custom gcode end: feature_gcode

;TYPE:Internal infill

```

The initial/starting extruder, always have the extrusion line when is triggered:
```
;-----Tool change g-code end-----------
; custom gcode end: toolchange_gcode
M106 S242.25 ; enable fan
T1 ; change extruder
G92 E0 ; reset extrusion distance
; custom gcode: start_filament_gcode
; Filament gcode
; custom gcode end: start_filament_gcode
G1 X116.462 Y116.462 F7200.000 ; move to first Internal infill point
G1 Z6.200000 ; restore layer Z
G1 E4.00000 F2400.000 ; unretract    ;<<<<<<<<<<<<<<<<<<<<<<<<<This line.
; custom gcode: feature_gcode
;-----Between extrusion role change g-code start-----------
;-----Between extrusion role change g-code end-----------
; custom gcode end: feature_gcode

;TYPE:Internal infill
```

**To Reproduce**
Steps to reproduce the behavior:
1. Load both stl and allow superslicer to merge both.
2. Assign the extruders to the parts
3. Print the result gcode.
4. See error

**>> Project File <<**
Stl files and gcode attached:
[DualExtruder.zip](https://github.com/supermerill/SuperSlicer/files/6528453/DualExtruder.zip)

**Expected behavior**
Add the missing extrusion command equal as the other extruder changes.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version: 2.3.55.5 / 2.3.56.0

**Additional context**
Nothing more to add, but please request me whatever you need.

Kind regards
Luis.

supermerill, 2021-05-24:
can I have your project file(s)?
Can't really replicate without those.

ldiegos, 2021-05-24:
Do you want the *.3mf?

supermerill, 2021-05-24:
yes

ldiegos, 2021-05-25:
Good morning supermerill,

Here you are:
[DualExtruderMissingExtrusionCommand.zip](https://github.com/supermerill/SuperSlicer/files/6536709/DualExtruderMissingExtrusionCommand.zip)

supermerill, 2021-06-09:
I just checked, but the first time it switch to the extruder, it doesn't need to unretract because it doesn't retract it before.

ldiegos, 2021-06-12:
But, without that command in there, the dual extruder/mixing nozzle, creates a blob in the printing after the lateral purge instead of the purge tower.

Could you please do something  to improve superslicer to the mixing printers?

Something like some kind of check in printer configuration to this type of printers? something like "purge bucket mixing nozzle"

It will be a great advantage for the geeetech users!

Luis.

supermerill, 2021-06-12:
you can add a very thin purge tower, it will allow the nozzles to fill up.

ldiegos, 2021-06-12:
> you can add a very thin purge tower, it will allow the nozzles to fill up and so be perfectly good for printing.
> You can also add a small patch on layer 1, to initialize the nozzle that don't print in the first layer.

Sorry supermerill, but this printers only have one nozzle, so can't do anything of this.  One nozzle, two extruders, one heated block that allow you to extrude the two filaments at the same time withing a mixing percentage. a U-type extrusion system.

Also, the purge tower, is generate on every layer all the time, so in my opinion, is a waste of filament. But i will give it a try without really wanting... lol ;)

Damm, nobody likes the bucket purge!!! i feel so sad!!! perhaps i need to create a python script like in cura, to analyze the g-code and add my own line when it needed.

Thanks so much for your great effort.

supermerill, 2021-06-12:
> add my own line when it needed.

you already add your line in the toolchange gcode 

> Sorry supermerill, but this printers only have one nozzle, **so can't do anything of this.**

Why ???? maybe the wording confuse you, replace "nozzle" by "filament/extruder"
....
[DualExtruderMissingExtrusionCommandBETTER.zip](https://github.com/supermerill/SuperSlicer/files/6642456/DualExtruderMissingExtrusionCommandBETTER.zip)

closed 2021-06-12 by supermerill

ldiegos, 2021-06-12:
> > add my own line when it needed.
> 
> you already add your line in the toolchange gcode. You can another one.
> 
> > Sorry supermerill, but this printers only have one nozzle, **so can't do anything of this.**
> 
> Why ???? maybe the wording confuse you, replace "nozzle" by "filament/extruder"
> ....
> [DualExtruderMissingExtrusionCommandBETTER.zip](https://github.com/supermerill/SuperSlicer/files/6642456/DualExtruderMissingExtrusionCommandBETTER.zip)
> 
> > Damm, nobody likes the bucket purge!!!
> 
> it outputs almost the same amount of filament, but as it's not tightly packed, it takes a very huge amount of space.

Lets my check what is the "shape-box" is the first time that i see that