supermerill/SuperSlicer #1172: [2.3.56] Multiple holes in perimeter

zztopper, 2021-05-23:
**Describe the bug**
Slicer leaves many holes in perimeter when slicing.

**To Reproduce**
Put a model on a plate and slice. 

**>> Project File <<**
[Motor_Frame_x1.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6528815/Motor_Frame_x1.3mf.zip)

**Expected behavior**
No holes in perimeter after slicing the model

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS 11.3.1
 - Version 2.3.56.0

**Additional context**
Seems to be fixed if min extrusion length set to zero.

supermerill, 2021-05-24:
can't reproduce

[image]

if you load this project file, it does the thing you have in the picture?

zztopper, 2021-05-24:
Hmm, not anymore, even if I set min extusion length back to 0.02, as it was before.
Can't reproduce either.
Some strange fluctuation, I guess you can close the issue

supermerill, 2021-05-24:
Can't reproduce neither with some random old versions.
I fixed some errors from this feature in the last release.
If there is still some here, it's for me a big priority because everything go trough that.

closed 2021-05-24 by supermerill

aka13-404, 2021-06-03:
[image]
[image]

The issue persists on the latest Nigthly windows x64 #333
Setting min extrude to 0 does not fix it.

supermerill, 2021-06-03:
@aka13-404 
can you post your project file? can't reproduce without.

aka13-404, 2021-06-03:
Just checked with stock configs, works good now. Must be something in the old configs, that breaks the slicing. 
Have attached the "broken" project files, but if it can not be reproduced with clean configs, 100% a config issue.

[Projects.zip](https://github.com/supermerill/SuperSlicer/files/6593812/Projects.zip)

supermerill, 2021-06-03:
@aka13-404

the gcode viewer is freaking out because it lacks the start gcode that is hidden inside the klipper macro.
Have to do something to prevent that.

aka13-404, 2021-06-03:
It's specifically the new "only custom Start G-Code" checkbbox, that is causing this, just checked.
The prescence of mm104 or m190 in the start gcode section does not do anything.

I will remain with the old bypass for now:
[image]