supermerill/SuperSlicer #1175: Print stops right after it started since 2.3.56

nicedevil007, 2021-05-24:
**Describe the bug**
After start of the print I get this error:

```
// Move exceeds maximum extrusion (7.901mm^2 vs 2.560mm^2)
// See the 'max_extrude_cross_section' config option for details
```

I imported the profile bundles from 2.3.55 to 2.3.56 and expected that it will work as it does before.

**To Reproduce**
Steps to reproduce the behavior:
Start a print after it is sliced with recent version.

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
Printer will start printing and not stop after a few seconds

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.56.0

**Additional context**
Add any other context about the problem here.

Xploder, 2021-05-24:
[image]
and have a G92 E0 at the end of your start g-code aswell as a G92 E0 in your Before layer change G-Code
[image]

[image]

nicedevil007, 2021-05-24:
Thank you, will try it out, G92 E0 was already in my start gcode but I didn't used relative E distances

Xploder, 2021-05-24:
> Thank you, will try it out, G92 E0 was already in my start gcode but I didn't used relative E distances

Makes sense, if you zero out the extrusion value at the end of your Start GCode but don't use relative distances in your slicer then the next move is going to be something ridiculously long that Klipper rightfully stops.

nicedevil007, 2021-05-24:
> Makes sense, if you zero out the extrusion value at the end of your Start GCode but don't use relative distances in your slicer then the next move is going to be something ridiculously long that [...]

ahhhh thank you for clearing this up :) it is working now!

closed 2021-05-24 by nicedevil007