supermerill/SuperSlicer #1176: Blurry bed SVG

qwewer0, 2021-05-24:
**Describe the bug**
Bed SVGs appears blurry on the latest version.
Default `ender3.svg` in this case.

**Expected behavior**
Bed SVGs should appear sharp, as it does on older versions (e.g. `Nigthly windows x64 #308`)

**Screenshots**
Left: `Nigthly windows x64 #308`
Right: `Nigthly windows x64 #313`
[image]

**Desktop:**
 - OS: Windows 10
 - Version: 2.3.56.0 (`Nigthly windows x64 #313`)

The issue might be connected to https://github.com/supermerill/SuperSlicer/issues/1163

JasonGuthrie, 2021-05-24:
I have the same issue with the Prusa bed image too.

supermerill, 2021-05-24:
there is a opengl32.dll that shouldn't be here (in the root directory). Delete it.

JasonGuthrie, 2021-05-24:
> there is a opengl32.dll that shouldn't be here (in the root directory). Delete it.

Aye that's fixed both the blurry build plate and low frame rates. Keep up the good work.

qwewer0, 2021-05-25:
> there is a opengl32.dll that shouldn't be here (in the root directory). Delete it.

That fixed the blurry SVG, thanks.

closed 2021-05-25 by qwewer0