supermerill/SuperSlicer #1177: [FR] "Inteligent" Width and Spacing

Sineos, 2021-05-24:
**Is your feature request related to a problem? Please describe.**
I print a lot of technical items. Especially on thinner walls, e.g. 2 or 3 perimeters, I try to design in a way to avoid gap fills. Additionally I play with the Width or Spacing setting until the gap fills would disappear.

**Describe the solution you'd like**
Intelligent / dynamic / magic setting of the Width / Spacing parameters on model basis to avoid gap fills

**Describe how it would work**
Rough process sketch:
1. For Width / Spacing set "optimization margins", e.g. +20% / -5%
2. Analyze the model and suggest Width / Spacing that is causing an absolute minimum of gap fills.

Potential improvement:
* Use "paint on" optimization, i.e. only optimize for a certain part of the model
* Find optimal Width / Spacing only in the marked area

This would be useful, e.g. when I want to avoid gap fills in thinner parts of the model but do not care about the more robust ones.

Benefits:
* Potentially better surface finish
* More robust prints
* Less print time

qwewer0, 2021-05-24:
@Sineos https://github.com/supermerill/SuperSlicer/issues/682#issuecomment-729661236 might be interesting to you, as it was for me.

Sineos, 2021-05-25:
Many thanks @qwewer0. Indeed interesting.

Now if SS could do some Width / Spacing tuning automagically, this would be cool. Example:

[image]

Left one sliced at 0.40 width and right one at 0.382

or
[image]

Left one sliced at 0.40 width and right one at 0.415

CCS86, 2021-05-25:
I definitely think this would be a useful feature, especially when you are printing somebody else's designs. 

It sounds tricky to implement.

For now, with the 2.3.56 release, you can design your "all-perimeter" areas to be a multiple of some reasonable spacing ie 0.4mm, then set widths like this:

[image]

[image]

supermerill, 2021-05-26:
> It sounds tricky to implement.

a bit. Maybe something like:
For each perimeter loop to create, for each point of it, launch a ray and get the distance it can travel before touching the other side of the loop. If the min is in the interval, use this value, or else the default one.

Sineos, 2021-05-26:
I'm surely completely clueless about the inner workings of a slicer, so just some random spitballing:
- Surely there are different "gap-fill-widths" possible in one model
- Minimize gap-fills globally, i.e. find a Width setting that is causing an absolute minimum over the whole model?
- Minimize gap-fills locally, i.e. find a Width setting that is causing no gap fill in a part or region of the model?
- Would different Widths in different parts of the model be possible?

supermerill, 2021-05-26:
> Would different Widths in different parts of the model be possible?

yes

> Surely there are different "gap-fill-widths" possible in one model

The width of the gap fill depends of the empty width at each point. See [vonoroi](https://www.boost.org/doc/libs/1_75_0/libs/polygon/doc/voronoi_main.htm)

> Minimize gap-fills globally, i.e. find a Width setting that is causing an absolute minimum over the whole model?
>Minimize gap-fills locally, i.e. find a Width setting that is causing no gap fill in a part or region of the model?

too costly, you don't want to multiply the time it takes to slice by ~100000

NameOfTheDragon, 2021-05-27:
I raised a related issue in #1085 which may go some way towards helping with this, although not completely. I understand this has been implemented - I haven't tested the fix yet but If @supermerill has implemented it how I envisioned it, and you know the width of walls used in the model, then the calculations become much simpler: desired wall width / number of perimeters = extrusion width.

supermerill, 2022-07-26:
I've added the "wall thickness".
You enter the wall thickness you used when designing your part, and it should adjust the perimeter width. to avoid gapfill.