supermerill/SuperSlicer #1178: [FR] Cardinal and Intercardinal Seam position

qwewer0, 2021-05-24:
**Describe the solution you'd like**
It would be nice to have seam position for all cardinal directions (`North`, `East`, `South`, `West`) and intercardinal directions (`North East`, `South East`, `South West`, `North West`) replacing the rear option

**Describe how it would work**
Seam would be placed at the selected direction, just like `Rear` (`North`) option right now

**Describe alternatives you've considered**
Alternatively the names could be `Rear` (which we already have), `Right`, `Front`, `Left` and `Rear Right`, `Front Right`, `Front Left`, `Rear Left`, if the direction names are confusing

qwewer0, 2021-05-24:
Maybe as an extra, a coordinate based seam positioning (like in cura) would be great addition.

Not sure how this should be done, but what I imagine is to take the parts coordinate as 0,0 and then we can add a coordinate pair e.g. 10,10 and along the vector from 0,0 to 10,10 the seam would be positioned.

But maybe with the seam painting that we already have this wouldn't be used.

supermerill, 2021-05-24:
why not just rotate the part?

qwewer0, 2021-05-25:
> why not just rotate the part?

It's not feasible when printing a test piece that needs to be in a specific orientation in X and Y, and sadly there are cooling ducts on printers that cools better from one direction, so in order to print some model well, the orientation needs to be selected thoughtfully.

cliffiard, 2021-05-30:
Hi. Sorry for posting in the wron area, but I'm intrested on how did you solved this issue https://github.com/MarlinFirmware/Marlin/issues/19650  and if you could help me out.
Thank you