supermerill/SuperSlicer #1179: A debugging check in this application has failed

bttguy, 2021-05-24:
Hi!

I downloaded the latest beta release 2.3.56 for win10. Upon opening I get these error msg-s. Than the slicer opens if I click on continue, but when I modify settings the same errors pop up, and after some time the slicer crashes and force close. Anything that I done wrong, or is this a bug? I used the previous releases with success.
[image]
[image]

tome9111991, 2021-05-24:
Look at
https://github.com/supermerill/SuperSlicer/issues/1166

bttguy, 2021-05-24:
Thanks @tome9111991. Just started to follow the project on github, missed that one. Closing this, and keep watching the mentioned issue :)

closed 2021-05-24 by bttguy