supermerill/SuperSlicer #1181: Overhang speed reduction and strange infill near overhang walls

MLiv79, 2021-05-24:
On overhang wall i get this strange infill that is unwanted, which setting make this?
[image]
[image]

And moreover i can't understand where this speed setting works on because on overhangs it doesn't reduce the speed. Moreover there is no setting to tells the angle where i'd like to consider a wall an overhang
[image]

supermerill, 2021-05-24:
bottom layers

supermerill, 2021-05-24:
> And moreover i can't understand where this speed setting works on because on overhangs it doesn't reduce the speed. Moreover there is no setting to tells the angle where i'd like to consider a wall an overhang

it's here (print->perimeters)
[image]
50% means 22.5°
100% means 45°
200% means 67.5°

MLiv79, 2021-05-24:
> bottom layers and/or top layers

Bottom and top layer are only 4 (0.2mm) but this material close to internal walls is until 6 or 7 mm in height. It seems to be related to overhangs because is only where the slope is to be considered an overhang. But as shown in the picture i attach in the first post, i've no options enable to add extra perimeters close to overhang wall.

MLiv79, 2021-05-25:
> > And moreover i can't understand where this speed setting works on because on overhangs it doesn't reduce the speed. Moreover there is no setting to tells the angle where i'd like to consider a wall an overhang
> 
> it's here (print->perimeters)
> [image]
> 50% means 22.5°
> 100% means 45°
> 200% means 67.5°

This work thanks. I don't know if it is my lack but reading the tooltip text passing over with the mouse it was impossible to understand that 50% was 22.5° 100% 45° and so on, because the help tooltip text talk about a % of the nozzle diameter...
Does exist somewhere a guide to read for these advanced (and very usefull) features?

supermerill, 2021-05-25:
> Does exist somewhere a guide to read for these advanced (and very usefull) features?

not made yet. Ideally, https://github.com/slic3r/Slic3r-Manual should be updated

closed 2021-06-04 by supermerill