supermerill/SuperSlicer #1184: Incorrect feedrate units when g-code flavor is Klipper?

balkin, 2021-05-26:
**Describe the bug**
I'm not 100% sure, but I believe the values in Printer Settings -> Machine Limits are in mm/s, not in mm/min

**To Reproduce**
Steps to reproduce the behavior:
1. Go to **Printer Settings** tab
2. In `General`, change G-code flavor to Klipper
3. Click on 'Machine limits'
4. Scroll down to 'Maximum Feedrates'
5. See maximum velocity in mm/min

**Expected behavior**
Units should be in mm/s and displayed in mm/s, not mm/min. Or maybe when switching from Marlin and 300 mm/s, these values should be converted to 18000 mm/min. I prefer mm/s though.

**Screenshots**
1. Marlin mode
[image]
2. Klipper mode
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.56.0

**Additional context**
In Marlin mode, it displays in mm/s and emits in mm/s:
```M203 X300 Y300 Z2 E120 ; sets maximum feedrates, mm/sec```
In Klipper mode, these values are ignored (`SET_VELOCITY_LIMIT` may be used though), but UI displays exactly same value 300 in mm/min.

supermerill, 2021-05-26:
Marlin is mm/s, reprap is mm/min ...
klipper is neither, so which one to choose?
If I change and you alternate between a 18000mm/min from a reprap flavor and a 18000mm/s klipper flavor, you can make the same complain.

I'm not again changing the unit, it's only used for the time calculation with klipper anyway.

NameOfTheDragon, 2021-05-27:
> I'm not again changing the unit, it's only used for the time calculation with klipper anyway.

With respect, that seems like the wrong decision that would cause a lot of confusion. I understand that there are a lot more important things to work on and its OK to make this low priority, but I do think it should be changed. Why not mark it "up for grabs" and see if anyone will take it on?

I suspect that most people changing to Klipper are coming from Marlin so will be familiar with mm/s. Moreover, speeds are almost always given in mm/s within the firmware, including in the Klipper configuration file, so regardless of where users are coming from, they will have to get used to mm/s when using Klipper. The justification that RepRap firmware uses mm/minute is a non-sequitur. This is not even consistent with Klipper itself, which uses mm/s everywhere that speed is referenced.

The use of mm/minute obviously comes from CNC mills and lathes where tool movement is typically an order of magnitude slower. This has carried through into G-code where it makes sense in that context. However, that makes no sense in slicers for 3D printers where toolhead movement is orders of magnitude faster. The units used in G-code shouldn't affect what is used by the end user. That's a bit like saying numeric value literals in C should be given in binary because that's what the CPU uses.

The mm/min unit is awkward for everyone and difficult to comprehend for 3D printer users. How can you visualize how far the nozzle will move in a minute? That could be getting on for 20 metres or more in a fast printer and you just can't visualize what that looks like. Millimeters per second is the right unit for 3D printing.

NameOfTheDragon, 2021-05-27:
> I'm not again changing the unit, it's only used for the time calculation with klipper anyway.

With respect, that seems like the wrong decision that would cause a lot of confusion. I understand that there are a lot more important things to work on and its OK to make this low priority, but I do think it should be changed. Why not mark it "up for grabs" and see if anyone will take it on?

I suspect that most people changing to Klipper are coming from Marlin so will be familiar with mm/s. Speed settings are almost always given in mm/s within the firmware, including in the Klipper configuration file. The justification that RepRap uses mm/minute is a non-sequitur. This is not even consistent with Klipper itself, which uses mm/s everywhere that speed is referenced.

The use of mm/minute obviously comes from CNC mills and lathes where tool movement is typically an order of magnitude slower. This has carried through into G-code where it makes sense in that context. However, that makes no sense for 3D printers where toolhead movement is orders of magnitude faster. The units used in G-code shouldn't affect what is used by the end user. That's a bit like saying numeric value literals in C should be given in binary because that's what the CPU uses.

The mm/min unit is awkward for everyone and difficult to comprehend for 3D printer users. How can you visualize how far the nozzle will move in a minute? That could be getting on for 20 metres or more, you just can't visualize what that looks like. Its not a great idea to be encouraging the use of mm/minute and I think mm/second should be preferred whenever possible.

balkin, 2021-05-27:
I think the proper behaviour is either to use mm/s everywhere or change 30 mm/s to 1800 mm/min then (automatically convert the value when changing the unit). But since it needs extra coding, I assume it's easier to just use mm/s everywhere. After all, printing speed is in mm/s even with Klipper variant of gcode, which is absolutely fine for me:
[image]

I do not know the exact source code history, and upstream doesn't have "klipper" variant at all so there's no other reference, but I'm just confused by the fact unit changes when I change g-code flavor but value doesn't.