supermerill/SuperSlicer #1185: Avoid nonsense curved bridging anchors/edges

jpedrick-athenacr, 2021-05-27:
**Is your feature request related to a problem? Please describe.**
When supporting a curved wall on top of infill, the slicer generates a curved anchor over the infill and then attaches the bridge infill to that anchor. The unsupported curved infill bridge anchor perimeter either pulls straight on short segments or sags on longer segments.

The result is that the following bridged infill pattern either:

- overshoots the bridge perimeter causing little loops that either sag, lift or tangle with their neighbors.
- undershoots and does not attach to the bridge perimeter.

This happens for me on most prints to one degree or another, usually it gets covered up by subsequent layers, but requires more top layers to overcome, where only one or two top layers may be desired.

**Describe the solution you'd like**

I think a few different solutions would resolve the problem, with various pros and cons for each.

1. On layers that require bridging infill, ensure that the bridging infill always attaches to an external perimeter. This means covering areas with bridging infill that would normally have a hole with a bridge infill perimeter. 
> Pros: this is likely the simplest to implement.
> Cons: when the bridging infill area would have been small and the void very large, this may be sub-optimal and significantly slow down the layer and use more filament.
2.  The slicer intelligently generates straight bridge anchor paths between infill supports and then attach the bridge infill to the straight anchor, for one or more layers.
> Pros: similar to current implementation and will avoid covering large interior voids with bridging infill when unnecessary.
> Cons: If bridging infill would have a lot of small voids( imagine a large lego with lots of pegs ), this could generate complex and slow bridging( solution #1 would be better ).

I would also like more control over the infill bridging in general.
Options like:
- only straight bridge segments where unsupported
- number of infill bridge perimeters
- infill bridge perimeter anchor overlap(how much to overshoot bridging infill over bridging perimeters)
- infill bridge perimeter overlap(how much each bridge perimeter overlaps with the previous bridge perimeter)

**Describe how it would work**
I think I've described that above.

**Describe alternatives you've considered**
Described above.

**Additional context**
[image]

supermerill, 2021-05-27:
Can you post your project file?

jpedrick, 2021-05-27:
@supermerill I've uploaded the project here: https://github.com/jpedrick/things/blob/main/ovular_planter_mold-inside_part.3mf
```

SuperSlicer
Version:   2.3.55.5
Build:     SuperSlicer_2.3.55.5

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.8.0-53-generic x86_64
Total RAM size [MB]: 33,581MB
OpenGL installation
GL version:   4.6 (Compatibility Profile) Mesa 20.2.6
Vendor:       X.Org
Renderer:     AMD VEGAM (DRM 3.38.0, 5.8.0-53-generic, LLVM 11.0.0)
GLSL version: 4.60
Installed extensions:

GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_multisample_advanced
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
[... 227 lines cut ...]
GL_NVX_gpu_memory_info
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_NV_vdpau_interop
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

```

jpedrick, 2021-06-10:
Hi @supermerill , I downloaded the latest release 2.3.56.3 AppImage and I'm still getting strange unsupported perimeters of internal infill layers

[image]

supermerill, 2021-06-10:
In your first project, I don't really see how to improve the dense infill more...

On the second one, what is strange? If you want more anchor for the bridge, increase the number in the infill tab.

jpedrick, 2021-06-10:
Hi @supermerill I would expect the solid infill layer to anchor to either a perimeter or the edge of the infill below it. Here it prints onto nothing. From my point of view printing onto thin air is undesired behavior and requiring dense infill for ordinary prints is also undesired behavior. I'm confused how this is "working as intended" when there aren't any FDM printers that would print this situation well. 

Is there a setting to:
a) print a bridged anchor for the infill to attach to?
b) print the entire layer with bridge infill so that the bridge infill attaches to perimeters?
c) ensure the bridge line endpoints attach to infill on the previous layer?

Thanks!

supermerill, 2021-06-10:
a) dense infill is trying to do that.
b) put 10000 in bridge anchor setting. but be careful, some geometries can create a bridge in every layer.
c) set your sparse fill density to 50%

closed 2021-06-14 by supermerill

moolight-seashell, 2023-05-02:
ok i thik i get it , i have this object to print : 
[image]

but when i slice it the firsts layers of the top cap look like that : 
[image]

and a bridge is form with hole, but there is no hole to properly form the bridge, on the last layers i hav this : 
[image]

and i will expect something like that for the first layer

moolight-seashell, 2023-05-03:
i think to solve this probleme the best wold probaly be to create e a new settings like "smallest inside hole allowed" in mm2 (probably intermediate or advanced setting)

and when a inside hole is smaller than the settings , the sliscer must be ingoring the hole making a full bridged surface.

in other hand, when the hole is equals ou greater than the settings, the sliscer generate inside perimeters for at least the first bridged layer