supermerill/SuperSlicer #1191: Build error in 2.3.56.0: Inherited from PrusaSlicer - Linker issue (DSO missing from command line)

TheAndr0id, 2021-05-27:
I'm sorry to do this, but this is another issue inherited from the 2.3.x PrusaSlicer source tree.  This issue hasn't been addressed yet, but the issue can be found here:

https://github.com/prusa3d/PrusaSlicer/issues/5816

Basically what seems to be happening is the libraries are being linked in the wrong order.  This results in a "error adding symbols: DSO missing from command line" error during the final linking of the SuperSlicer binaries.

I know, what the?  DSO missing from command line?  What kind of fresh hell is this now?

\me rolls up sleeves, fires up google.

So again after much digging (this pandemic is good for something) I found out about the magic linker option:
`--copy-dt-needed-entries`

If you check out the linked PrusaSlicer issue you can read all about what I found out.

Possible fixes:

1)  Determine why the message is happening and reorder the libraries as they are added to the command line for the linker.  From [TFA](https://stackoverflow.com/questions/19901934/libpthread-so-0-error-adding-symbols-dso-missing-from-command-line):

 > You should solve this problem by making sure that all libraries that are needed to satisfy the needed symbols are directly specified on the linker command line. Also keep in mind that order often matters.

2)  Add the following into your LDFLAGS:   -Wl,--copy-dt-needed-entries
      I do this via the cmake call during the build process:

```
cd build
LDFLAGS="-Wl,--copy-dt-needed-entries" cmake -DCMAKE_BUILD_TYPE=Release .....
```
Ok, so the second isn't a fix, it's a work around, but it works.

This issue *might* be fixed with PrusaSlicer 2.4.x, but it's still going to cause issues with building the 2.3.56.x branch of SuperSlicer.  If anything this documents this problem with the work around...

supermerill, 2021-05-27:
> What distro? What Linux?

> It doesn't matter

it matters because I have no problem building on ubuntu 18.04

TheAndr0id, 2021-05-27:
There might be a couple of things in play here.  I'm using Slackware x64 with gcc 9.2.0.  Gcc 9 is a bit more hinky than my previous release (5.5) and generates a LOT more warning messages.  This might be part of the issue.

Second, Ubuntu provides 32bit backwards compatibility with their installs (or they did, it's been a while since I've looked at Ubuntu).  This means extra libraries, includes and compiler options installed.  They also have many cross links between 32bit and 64bit shared libraries to make their own compiled binaries behave.  Ubuntu is also what I like to call a fat distro.  They install EVERYTHING including the kitchen sink.  :)

Slackware x64 is 64 bit ONLY.  There's no compatibility for 32 bit stuff (thus the lib64 directories).  If code is a little "loose" (please don't take that as any sort of insult) with the library resolution then the builds need some extra help.  (for instance assuming /usr/lib and not looking for /usr/lib64)  

Many of the packages I've had to build since I started with 3D printing (OpenSCAD, CGAL, Meshlab) have need some extra attention to play well with strict x64 installs.  Some can be fixed with extra options, some require "weird" linker options as above.  From the description of the option in the man page it "reads to me" as a work around for something not so good.

GCC is also a bit of a moving target as well.  Their releases are scattered all over (8.5 was just released, while 11.1 was released a month ago) and I'm totally confused.  9.2 was released in 08-2019 so it should be new enough...

I'm not a developer, I'm a sysadmin and I ensure I document my build/install processes as I go.  When I hit bumps in the build process I look into them and if it's something that seems wonky I try to point it out.

The reason I stated the distro doesn't matter because the distro doesn't impact how the compiler/build tools run.  I have to assume that the maintainers of all the distributions follow the correct process for making the compiler/linker/libraries land in the right place.  If something is not installed, then that's a distro problem, and in this case it's not the issue.

I did a little more digging and without installing Ubuntu myself it seems that gcc 7.x is the default version.  Maybe this is another gcc 9.2 thing?  Kind of a "hi, this is gcc from the future and I'm going to break your sh!t..."?

supermerill, 2021-05-27:
Also, I don't know how you build.
In the current build process, every lib has its source files downloaded and built and then statically linked, to avoid these kinds of problems.

TheAndr0id, 2021-05-27:
Here's exactly my build process (in it's current state):

First edit deps/wxWidgets/wxWidgets.cmake add in -DwxUSE_MEDIACTRL=OFF

(this is run from inside the base directory SuperSlicer-<version>)
```
mkdir build deps/build
( cd deps/build ; cmake -DOpenGL_GL_PREFERENCE=LEGACY .. ; make )

cd build
LDFLAGS="-Wl,--copy-dt-needed-entries" \
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/SuperSlicer  \
      -DCMAKE_PREFIX_PATH="`pwd`/../deps/build/destdir/usr/local" \
      -DSLIC3R_STATIC=1 -DSLIC3R_WX_STABLE=1 \
      -DOpenGL_GL_PREFERENCE=LEGACY -DSLIC3R_BUILD_TESTS=0 ..
make
make install
```

This is pretty much following the build instructions from your/PrusaSlicer FAQ.

supermerill, 2021-05-28:
As i'musing gtk2, i need to do a `cp libwxscintilla-3.1.a libwx_gtk2u_scintilla-3.1.a`
Also i don't use `DSLIC3R_WX_STABLE` (and it's not in my doc) because I need version 3.1
I also don't use any LDFLAGS, and I don't mess with opengl, but maybe it's mandatory in your distro.

good luck. I can't help more.

TheAndr0id, 2021-05-28:
Thanks for the help!

Just to clarify, it looks like the DSLIC3R_WX_STABLE now pulls in the 3.1 from the Prusa dep repo as part of the make dep.  The LDFLAGS are just for the compiler flag mentioned in the issue for the DSO missing error.  The OpenGL thing isn't needed any more, the Prusa guys fixed that warning/error (it was causing a *large* warning message to spew out about not setting the default OpenGL library type and that it defaulted to LEGACY which is what the code needed in any case.  (The closed Prusa issue is here https://github.com/prusa3d/PrusaSlicer/issues/4894)