supermerill/SuperSlicer #1192: Artifact around a box

Liger0, 2021-05-27:
**Describe the bug**
There's a weird artifact around a box (not translation related)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Printer settings -> General
2. Look at the thumbnail category

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
no artifact?

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows10
 - Version Better spacing and others

**Additional context**
Add any other context about the problem here.

csch0, 2021-05-30:
Its fixed now with this commit: https://github.com/supermerill/SuperSlicer/commit/3ed01dad9cce4035cbd71c3d03363955867a12e4

Nice work, was just about to fix the same.

supermerill, 2021-05-30:
it was the 'big:' label that was misplaced.
fixed some other bugs, like missing "reset"

closed 2021-05-31 by Liger0