supermerill/SuperSlicer #1193: Question: Way to control retracted vs un-retracted travels?

CCS86, 2021-05-27:
SS seems to do both retracted travels and un-retracted travels (combing in Cura). But I can't find any settings that govern which is used, and where. Examples:

- Maximum length of un-retracted travels
- Where are un-retracted travels allowed? (1st layer, solid infill, sparse infill)
- Ooze compensation for un-retracted travels (ideally based on distance or time)

Do any of these controls exist, or would all of these be feature requests?

supermerill, 2021-05-27:
> Maximum length of un-retracted travels

minimum length after retraction

> Where are un-retracted travels allowed? (1st layer, solid infill, sparse infill)

everywhere

> Ooze compensation for un-retracted travels (ideally based on distance or time)

extra length on restart

CCS86, 2021-05-27:
> > Maximum length of un-retracted travels
> 
> minimum length after retraction

Isn't this the minimum length of print move required after a retraction, before another retraction is allowed? 

That would be quite a bit different from the "maximum length of un-retracted travel"

> > Where are un-retracted travels allowed? (1st layer, solid infill, sparse infill)
> 
> everywhere

This could be a useful setting to have.

> > Ooze compensation for un-retracted travels (ideally based on distance or time)
> 
> extra length on restart

Again, correct me if I am wrong, but this is an extra length applied to the un-retraction move. Since un-retracted travels don't have this move, I assume it wouldn't be applied.

Ideally this is never used because retraction should be well tuned to not lose filament. But with un-retracted travels, ooze is unavoidable, especially on a bowden printer.

supermerill, 2021-05-27:
> Isn't this the minimum length of print move required after a retraction, before another retraction is allowed?

the tooltip say: `Retraction is not triggered when travel moves are shorter than this length.`

> This could be a useful setting to have.

you can add a modifier, for the first layer for example. Automated modifiers aren't here yet.

> Again, correct me if I am wrong, but this is an extra length applied to the un-retraction move. Since un-retracted travels don't have this move, I assume it wouldn't be applied.

Oh, yes. As not-retracted moves are only short (because of `minimum length after retraction` setting) there wasn't any need for that. This setting is still a stop-gap.

CCS86, 2021-05-27:
> > Isn't this the minimum length of print move required after a retraction, before another retraction is allowed?
> 
> the tooltip say: `Retraction is not triggered when travel moves are shorter than this length.`

Maybe the tooltip is wrong? Or does this same setting behave differently in SS?

https://manual.slic3r.org/expert-mode/fighting-ooze#:~:text=Minimum%20travel%20after%20retraction%20%2D%20Triggering,move%20before%20considering%20a%20retraction.

> > Again, correct me if I am wrong, but this is an extra length applied to the un-retraction move. Since un-retracted travels don't have this move, I assume it wouldn't be applied.
> 
> Oh, yes. My bad. As not-retracted moves are only short (because of `minimum length after retraction` setting) there wasn't any need for that. This setting is still a stop-gap.

Depending on the answer about the setting above, there might be a good use for a sister-setting for un-retracted travels, ideally as a length of travel multiplier.

CCS86, 2021-05-28:
I'm pretty sure that tool tip is wrong. I have minimum length after retraction set to 2mm and it is doing easily 20mm long un-retracted travels.

supermerill, 2021-05-28:
> Maybe the tooltip is wrong? Or does this same setting behave differently in SS?

Weird, as this tooltip was created by the original slic3r author in 2015. I'll dive into the code to verify, but it should be good.

CCS86, 2021-05-28:
Check this layer out. That is a **ton** of un-retracted travel:

[image]

CCS86, 2021-05-31:
Well shoot. It appears that the "only retract when crossing perimeters" setting was enabled and causing this behavior. With that disabled, the "minimum length after retraction" setting, while misleadingly named, behaves like the tool tip states. 

I think a more accurate name would be "maximum length for un-retracted travel"; or "minimum travel length to trigger retraction". 

 Cura has a similarly named setting that behaves as the name suggests:

[image]

closed 2021-05-31 by CCS86