supermerill/SuperSlicer #1194: Install Issue 2.3.56.1 - make install fails (missing .desktop files)

TheAndr0id, 2021-05-28:
This involves release 2.3.56.1, under Linux x64 (Slackware).

During the final build step using a "make install" to install the final built binaries into (in my case) /opt/SuperSlicer the following error occurs:  (I've removed my home directory from the output for privacy reasons)
```
[100%] Built target Shiny
Install the project...
-- Install configuration: "Release"
CMake Error at cmake_install.cmake:54 (file):
  file INSTALL cannot find
  ".../SuperSlicer-2.3.56.1/src/platform/unix/Slic3r.desktop":
  No such file or directory.
```
And it's correct, there are no .desktop files under the src/platform/unix directory.  Only .desktop.in files.

I build this under a build directory, so shouldn't the install  be looking for the .desktop files under the build/src/platform/unix directory?  In any case there are no .desktop files there either.

I copied some .desktop files into the src/platform/unix directory and the install completed.  

I'm guessing there's a missing make process to take the .desktop.in files and fill in the details to make them .desktop files.  Those resulting .desktop files should be placed under the build directory and the cmake_install.cmake file should be using that directory for the .desktop installs.

supermerill, 2021-05-28:
The .desktop is made at the ./cmakelists.txt:529
it should be in build/src, and be named superslicer.desktop

but I just see that the `else` below miss the change, I'll modify that.

If you want to try it yourself now, replace the code in the `else ` line 535 by
```

    configure_file(${LIBDIR}/platform/unix/Slic3r.desktop.in ${LIBDIR_BIN}/${SLIC3R_APP_KEY}.desktop @ONLY)
    configure_file(${LIBDIR}/platform/unix/Gcodeviewer.desktop.in ${LIBDIR_BIN}/${SLIC3R_APP_KEY}-Gcodeviewer.desktop @ONLY)
    install(FILES ${LIBDIR_BIN}/${SLIC3R_APP_KEY}.desktop DESTINATION ${CMAKE_INSTALL_PREFIX}/resources/applications)
    install(FILES ${LIBDIR_BIN}/${SLIC3R_APP_KEY}-Gcodeviewer.desktop DESTINATION ${CMAKE_INSTALL_PREFIX}/resources/applications)
    install(DIRECTORY "${SLIC3R_RESOURCES_DIR}/" DESTINATION "${CMAKE_INSTALL_PREFIX}/resources")

```

TheAndr0id, 2021-05-28:
That worked correctly.  The .desktop files install as they should.  They are also under the build/src directory, not under build/src/platform/unix:

```
> ls SuperSlicer-2.3.56.1/build/src/*.desktop
SuperSlicer-2.3.56.1/build/src/SuperSlicer-Gcodeviewer.desktop
SuperSlicer-2.3.56.1/build/src/SuperSlicer.desktop
```
I don't know if that's what you were wanting but I now have a complete build.  Ready to go for the release version!
:D

closed 2021-06-13 by supermerill