supermerill/SuperSlicer #1195: Single nozzle, Multi Materials Klipper Gcode Issue

kackle1998, 2021-05-28:
SuperSlicer_2.2.53
Windows 10 64-bit

I'm running a custom I3 style machine on Klipper that uses a "false extruder" of sorts, where I run multiple materials through the same stepper motor. I understand that makes very little sense, so it's easier to show. Here is someone else's setup using the same principle for two materials. https://www.youtube.com/watch?v=0F38WQUEaGs&t=2s Mine uses eight.

When slicing, I set up each color of my filament as a different filament, and assign each to their own extruder. The problem is, all of these are technically running under the same motor, so the line `ACTIVATE_EXTRUDER EXTRUDER=extruder2` causes a print failure, as I truly only have one motor. 

If the line `ACTIVATE_EXTRUDER EXTRUDER=extruder ` could use the same convention as Marlin "T0, T1, T2" etc..., I could easily get around this using an empty Klipper macro, such as the following:
```
[gcode_macro T0]
gcode:
```
This would also be easily solved with an option within the SuperSlicer extruder tab under Printer Settings that allows me to link the false Extruder 2,3, and so on to what klipper knows as "Extruder." "

I noticed a commit that referenced an extremely similar issue here, https://github.com/supermerill/SuperSlicer/issues/276, but don't see the fix anywhere in SS. If I am just missing an option, could someone direct me to it? Thank you

supermerill, 2021-05-28:
I'll trash the current check and will replace it by "if there is something, I'll output nothing"

Roycinger, 2022-12-03:
Is this still ongoing? I'm basically still running into the same issue