supermerill/SuperSlicer #1197: Source Package Issue?  (2.3.56.1)

TheAndr0id, 2021-05-28:
This may or may not be an issue - maybe more of a question.  This is with version 2.3.56.1, I'm building under x64 Linux, but that isn't really important...

I've been doing build prep in anticipation of a new stable release of the 2.3 branch.  The build and install issues have been worked out and so far, so good.  Next, I move on to testing and migrating my 2.2 configuration.

There are no profiles installed.  What?  Now what happened....

I take a look and surprise!  There's nothing in the resources/profiles directory in the source tarball.

The source bundles do not include any of the printer profiles.  Is this because they are a pull from the Slic3r git repo and aren't included in guthub's auto source tarball pull?  Is it because this is a pre-release version?  Is this a mistake?   Is there a new process to pull in the contents of the profiles directory?

I can go through the process of pulling the download myself via a zip file, but that's a bit of a pain.  Is there a make target to pull the profiles git repo down into the source tree so it can be installed?  How can I be sure I'm pulling the right profiles versions that match with the SuperSlicer version?

For now I'm going to do the zip pull from github to get me moving forward, but having a valid non-manual process moving forward would be a good thing.

supermerill, 2021-05-28:
from the build doc https://github.com/supermerill/SuperSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md:

### How to get the source code

You have to gitclone  the repository
```
git clone https://github.com/supermerill/SuperSlicer.git
```

and then you have to clone the profiles submodules

```
cd resources/profiles
git submodule update
```

TheAndr0id, 2021-05-28:
I can understand that if you want to clone/use git to manage the software or make changes.  I'm not interested in using the current head of the development tree, just the released versions.  

I know enough code to know I shouldn't be wading into the development of a slicer...  :)

I'm using the download source tar.gz from the releases page.  This one:
https://github.com/supermerill/SuperSlicer/archive/refs/tags/2.3.56.1.tar.gz

I realize this tar ball is auto-generated by github (been down this road before).  This will cause headaches for distro maintainers who build their own releases of software as the source tarballs available for download aren't "technically" complete since they are missing the profiles.  Since they can't know all aspects of all the software they package there's a potential for a lot of packaged SuperSlicer software that doesn't work - this creates more headaches for you as people log issues...

Some maintainers build their own source tarballs by doing a git "export" (clone then remove all the .git files/directories) for their releases.  I'm not up to speed with all things github, but I believe it can be added to the automated tag builds for releases.

I'm not that concerned with the current 2.3.56.1 release, I'm just getting things lined up for stable.  Since your profiles will change as you move forward in development I'm going to assume you will need to include a "snapshot-in-time" of the profiles to include with the stable release.  If I go to grab the stable release source later, using git to grab the current profiles will introduce all worlds of hell...

For example:
Say I'm building 2.3.56.1.  Current is 2.3.56.5, but it's broken because, well, development.  I grab the profiles via git and I get the profiles that work with 2.3.56.5.  Those may (or may not) be compatible with 2.3.56.1.  I know from experience that Prusa/Super slicer crashes if the profile file is not in the correct format.  Now I'm diving down the rabbit hole trying to figure out what went wrong with my build when it turns out it's incompatible profiles....

In the short run, I've pulled the profiles via github to do my testing of 2.3.56.1.  If needed I'll download the Linux binary release alongside the source tree and copy over the profiles from the binary release into the source tree before I build.  It's just a bunch of extra steps in a process that is already getting pretty complicated.

supermerill, 2021-05-29:
basically, what it means it that the profiles are at https://github.com/slic3r/slic3r-profiles
You can also copy-paste them from any release zip if you don't want to use git.

TheAndr0id, 2021-05-29:
Yup, totally get that.  

The issue is how are you going to tag the version of the profiles to match the version of the slicer?  For instance what would the new extrusion width spacing settings do to a previous release?  (Overhang thresholds/Ironing settings are another example...)

supermerill, 2021-05-29:
profiles have a "min version" setting. The slicer won't load profiles that have a 'min version' higher than its.

closed 2021-06-14 by supermerill