supermerill/SuperSlicer #1198: default_print_profile not being selected on start

TheAndr0id, 2021-05-28:
**Describe the bug**
the default_print_profile value in the configuration file is set correctly and works on previous releases.  In 2.3.56.1 the print profile selected is the first available.  (In my case, 0.08mm HIGHDETAIL instead of the 0.20mm NORMAL)

This is the standard profiles as downloaded via the Slic3r git link.

**To Reproduce**
Open SuperSlicer, select a printer.  The print settings profile selected is always the first in the list not the default print profile.

The default_print_profile is set correctly as it is showing on the Printer Settings/Dependencies page in settings.

If a printer is selected and a print settings chosen closing the slicer and restarting *does* return the settings to their previous selections.  This is good.  

**Expected behavior**
When the slicer is opened it should select the default print profile as specified in the configuration file.  If the printer is changed the print settings should change to the default_print_profile.

This might need some clarification around switching when there's a sliced model on the platter or changes have been made in the settings.

In any case it doesn't look like the default_print_profile setting is doing anything at the moment.

**Desktop (please complete the following information):**
 - OS: Linux x64 (Slackware)
 - Version: 2.3.56.1
 
This is a build from source.

supermerill, 2021-05-28:
I installed the prusa mk3S profile via the wizard.
I select it
I see "0.15mm Quality" as the selected print. And it's the default_print_profile in the conf

TheAndr0id, 2021-05-28:
So I just tried the same thing.  Here's my total process:

1) rm -rf ~/.config/SuperSlicer ; /opt/SuperSlicer/bin/superslicer
2) Click through the two warning pop-ups (empty config directory and SSL certs)
3) When the wizard opens add a MK3S (0.4 nozzle)

One important note, this only happens when the config directory has been cleared.  If I select a print profile and exit it is retained on restart.

I've attached screenshots of what I get.
[image]
[image]

I also just tried this with PrusaSlicer 2.3.1 and the results were identical.  If it is something with the program and not my build (which is very possible) it is upstream.

I also tried the same action with the 2.2.53.4 release.  It selected the default print profile correctly.  Other than some tweaks due to the 2.3.0 code base (as mentioned in other tickets) these two binaries are built using the same process, libraries and equipment.  In fact my machine hasn't been rebooted since 2.2.53.4 was built.  (Unix for the win!  :P )

TheAndr0id, 2021-05-29:
Just another test I thought I'd try.  So I took and added a new printer to an existing config.  (I use a Ender 3V2)

If I select a print setting that the new printer doesn't have then when I switch to the printer it selects the default print profile.  So:

1) Ender 3v2 selected, print profile 0.2mm NORMAL selected.
2) Add Prusa MK3S.  It selects the 0.2mm QUALITY profile to match what we already had selected.
3) Change the profile to 0.15mm QUALITY profile from the MK3S config.
4) Change the printer back to Ender 3V2 and it switches to the default profile of 0.2mm NORMAL since the Ender 3V2 can't do 0.15mm.  (it does 0.16mm to match it's Zstep resolution)

When you switch to a print profile it stays selected even across application restarts.  It only shows up on a completely empty configuration (remove the ~/.config/SuperSlicer directory.  Once a print setting is picked SuperSlicer remembers it correctly and restores the settings on the next start.  It remembers it's settings when changing printers as long as the new printer also has a layer height that matches.  If there isn't a layer height that matches it switches to the default_print_profile as expected.

I now know and understand how Super/Prusa slicers choose their print profiles.  I didn't really want to know, but now I do.

This is a very low priority bug.

supermerill, 2021-05-29:
I changed some things to make it works
can you test it? I fear some regression.
you can fetch the source on the debug_ubuntu branch, or brag the debug_ubuntu build from the action Tab.

TheAndr0id, 2021-05-30:
I tried grabbing the two files involved in the change and added them to 2.3.56.1.  I hoped it was a quick way to do a test.  It failed horribly:

> 
> [2021-05-30 00:55:28.189073] [0x00007f967f2e5780] [error]   Uncaught exception: Unknown option exception: perimeter_round_corners
> Unhandled unknown exception; terminating the application.

So I did some down and dirty and edited Preset.cpp and removed the `perimeter_round_corners` to get a test.

The results look good, I did the following:

1) Deleted the ~/.config/SuperSlicer directory
2) Started SuperSlicer, added the Prusa MK3S.  It came up with the 0.15mm QUALITY profile selected. (correct)
3) Added in the Ender 3V2.  It selected the 0.20mm QUALITY profile since it doesn't have a 0.15. (correct)
4) Switched back to Prusa MK3S.  It switched to the 0.20mm profile for the MK3S. (correct/makes sense)

Other than the `perimeter_round_corners` not being valid with 2.3.56.1, it looks to work as *I* was expecting.

closed 2021-06-13 by supermerill