supermerill/SuperSlicer #1199: Support interface generated midair

qwewer0, 2021-05-29:
**Describe the bug**
Support interface generated midair without any support base bellow it.

**>> Project File <<**
[overhang-test.zip](https://github.com/supermerill/SuperSlicer/files/6564561/overhang-test.zip)

**Expected behavior**
Support interface should have a support base.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.1

supermerill, 2021-05-29:
>< Not now!

neophyl, 2021-05-30:
Given this is the same in PS 2.3 and that the PS 2.4 release when it happens is supposed to be targeting a lot of support issues I would wait until merging with 2.4.  No point try to fix code that will be updated and changing anyway as that sounds like a lot of wasted effort lol.

ashleyuk90, 2021-06-07:
Was told to move my issue here from 1272, just checked the stl in the ZIP of this issue against PS 2.3.1 and the supports are fine. So the issue is defiantly for both the attached and my example within some code of SS and not what was merged from PS 2.3.1 recently, or may be merged from 2.4.0 unless it would overwrite any custom SS support code?

supermerill, 2021-06-07:
There is definitely an issue in a change I made last year in the support code.

supermerill, 2021-06-25:
As expected, it was painful.
it's a difference of 0.00000001 mm in a variable that makes the rectilinear miss the threshold because the float is floored a little bit too early.

qwewer0, 2021-06-26:
Fixed. Tested it on the latest release.

closed 2021-06-26 by qwewer0