supermerill/SuperSlicer #1200: Feature Request: Dual Monitor Support / Turn Tabs into windows option

JasonGuthrie, 2021-05-29:
**Is your feature request related to a problem? Please describe.**
Yes. As I work with multiple monitors (and I assume many others do as they are becoming the norm) it is a small pain to have to flick between the Plater Tab and the other tabs when making incremental changes.

**Describe the solution you'd like**
If dual monitor support was created it would allow for faster and smoother workflows, as the constant tab switching would be vastly reduced. 

Therefore, if a preference option was given to split the Plater (or all tabs) into separate windows it would allow the user to breakout and arrange SS to suit their workflow.

supermerill, 2021-05-29:
Currently, all plater tabs are in fact the same thing that is 'moved' from tab to tab. Have to change a lot of the gui to change that.
I don't have the time for that and it may create more problems merging future version of prusaslicer.

closed 2021-05-30 by JasonGuthrie

JasonGuthrie, 2021-05-30:
> note: you can move the settings in their own window, the option is in the preferences -> gui.

Ah I was unaware of this option. This resolves the issue.

Thanks once again.