supermerill/SuperSlicer #1201: Gcode preview is missing lines

rsghosh, 2021-05-30:
**Describe the bug**
If both start_gcode_manual and relative_e_distances are set to 1, the gcode preview is missing line segments. However, the sliced preview looks fine. The problem only occurs if both of these are turned on, turning either off makes it disappear.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to printer settings > custom G-code, and check "Only custom Start G-code"
2. Go to printer settings > General, and check "Use relative E distances"
3. Slice a model and look at the Gcode preview

**>> Project File <<**
[calicat.zip](https://github.com/supermerill/SuperSlicer/files/6565651/calicat.zip)

**Expected behavior**
Turning on start_gcode_manual and relative_e_distances should not affect the gcode preview.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.3.56.1

supermerill, 2021-05-30:
You forgot to add `M83 ; use relative distances for extrusion` to your start gcode.
If you want it to be added automatically, add `{if use_relative_e_distances}M83 ; use relative distances for extrusion{endif}`

if you select "Only custom Start G-code" be **sure** to put in your custom start gcode all the required init command!

rsghosh, 2021-05-30:
Ah. I normally use a klipper macro for my start gcode and put `print_start PA=0.0426 EXTRUDER=[first_layer_temperature] BED=[first_layer_bed_temperature]` in the start_gcode, and the macro calls M83. I'll make sure to put M83 before print_start then so the slicer sees it. Thanks!

closed 2021-05-30 by rsghosh