supermerill/SuperSlicer #1203: Missing language template file

sziga, 2021-05-30:
The language template file (Slic3r.pot) does not contain many of the terms used in the program (e.g. Multimetarial, Vertical shells, Skirt & Brim, Cooling) and therefore cannot be translated in full.

supermerill, 2021-05-30:
That's because these texts aren't in the source code.
These ones should be retrieved by the pom_merger script.

sziga, 2021-05-31:
Thank you, now the terms are complete.
However, some terms are still not in the .po file, I will check where they are and add them manually.
These are the missing terms:

#: src/libslic3r/PrintConfig.cpp:2601
msgid "Fan PWM from 0-100"
msgstr ""

#: ../ui_layout/print.ui src/libslic3r/Config.cpp:30
msgid "Perimeters & Shell"
msgstr ""

#: ../ui_layout/print.ui src/libslic3r/Config.cpp:34
msgid "Skirt & Brim"
msgstr ""

#: ../ui_layout/print.ui src/libslic3r/Config.cpp:36
msgid "Width & Flow"
msgstr ""

#: ../ui_layout/print.ui src/libslic3r/Config.cpp:38
msgid "Multiple extruders"
msgstr ""

#: ../ui_layout/filament.ui src/libslic3r/Config.cpp:43
msgid "Cooling"
msgstr ""

#: ../ui_layout/filament.ui src/libslic3r/Config.cpp:50
msgid "Multimaterial"
msgstr ""

supermerill, 2021-05-31:
I see.
This will be fixed in the next nightly.

closed 2021-06-13 by supermerill