supermerill/SuperSlicer #1206: Gap fill on 2.3 increases generated gcode size dramatically.

ManuelMcLure, 2021-05-30:
**Describe the bug**
Gap fill on 2.3 increases generated gcode size dramatically.

**To Reproduce**
Steps to reproduce the behavior:
1. Slice the attached project on 2.2.53 and 2.3.56.1
2. Export the gcode
3. The gcode generated by 2.2 is 15MB, on 2.3 it is 35MB. Turning gap fill off on 2.3 reduces the size of the generated gcode to 15MB again, which means that the generated gap fill was 20MB.

**>> Project File <<**
[GcodeSizeIssue.zip](https://github.com/supermerill/SuperSlicer/files/6566955/GcodeSizeIssue.zip)

**Expected behavior**
Gap fill should not increase the gcode size that dramatically.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 2.3.56.1

**Additional context**
Opening per conversation with @supermerill on Discord.

CCS86, 2021-05-30:
What is your setting for:

Printer settings > general > advanced > minimum extrusion length

ManuelMcLure, 2021-05-31:
It's at the default of 0.035. I tried increasing it to 0.1 and it had hardly any effect on the resulting gcode size. I think I ended up at 30GB with that setting. Changing Print Settings->Filtering->Resolution from the default of 0 to 0.1 managed to reduce the size to 21MB.

CCS86, 2021-05-31:
Try it at 0.3

ManuelMcLure, 2021-05-31:
I already had a conversation with merill on Discord and he agrees that the gap fill is being too large and agreed for me to open an issue.

supermerill, 2021-05-31:
I think there is ways to increase slicing speed by adding a not-external perimeter precision.

CCS86, 2021-05-31:
> I already had a conversation with merill on Discord and he agrees that the gap fill is being too large and agreed for me to open an issue.

Did you try?

CCS86, 2021-05-31:
> I think there is ways to increase slicing speed by adding a not-'external perimeter' precision.

I'm sure you could, but non-external perimeters tend to "print through" to the outside. 

After we talked in my recent issue, which included incredibly dense gap filling moves, and you mentioned changing the way "minimum extrusion length" was applied, I was able to resolve the dense code with that setting. I think a lot of people will have profiles set wayyyy too small. I went from 0.006 to 0.3, a 50x change, with a similarly dense output. Might be worth forcing a change to that setting, or at least a very stern warning.

If anything, a "gap fill minimum segment length" setting would be a nice way to fine tune the resolution of gap fill, without affecting perimeters.

supermerill, 2021-05-31:
I should make some new test on my cheap ender3 to see what's the right default settings to avoid stalls.

ManuelMcLure, 2021-05-31:
> > I already had a conversation with merill on Discord and he agrees that the gap fill is being too large and agreed for me to open an issue.
> 
> Did you try?

I tried with 0.3 minimum extrusion length - 2.2 results in 14MB gcode and 2.3 still ends up at 22MB, so we still have a size increase of 8MB.