supermerill/SuperSlicer #1207: Time Estimates increase, even if you change settings that should make them decrease

nirurin, 2021-05-30:
**Describe the bug**
I have a print set up, that has a time estimate at 15 hours (for example). 
The print settings use 3 perimeters. 
I change to 4 perimeters, and the print time goes up to 18 hours (fine, makes sense). 
I change back to 3 perimeters, and the print time goes up to 21 hours (what?)
I change down to 2 perimeters, and the print time goes up to 23 hours (err)

Basically just keeps going up n up. Unless I delete the model / shut the program and start over. 

**To Reproduce**
Steps to reproduce the behavior:
Do as above. 

**>> Project File <<**
https://u.pcloud.link/publink/show?code=XZbkK9XZC1ebISYlI65soVckTrpDOkB5au7X

**Expected behavior**
Not this.

**Desktop (please complete the following information):**
Windows 10, newest version.

supermerill, 2021-05-30:
ironing post-process doesn't clean up as it should.

nirurin, 2021-05-31:
oh, so is it ironing specifically that messes up the calculations and keeps making them stack? That would make sense, as I didn't notice this problem before, and I started using ironing recently-ish

closed 2021-06-13 by supermerill