supermerill/SuperSlicer #1209: Display of Creality models is incorrect

sziga, 2021-05-31:
Only 4 Creality models are displayed in the configuration wizard.
Missing e.g. Ender3V2, Ender5, Ender2 etc.
They are listed in the profile file (Creality.ini).
For the models that are displayed, the images are not displayed.
[image]

supermerill, 2021-06-01:
can't reproduce
what is the version?

sziga, 2021-06-01:
version: SuperSlicer_2.3.56.1_win64_210527

supermerill, 2021-06-01:
can't reproduce on a fresh install
[image]

sziga, 2021-06-02:
Deleting the file C:\Users\ \AppData\Roaming\SuperSlicer\vendor\Creality.ini solved the problem.
A file version 0.2.5 was in the directory.
Thanks for your help!

closed 2021-06-02 by sziga