supermerill/SuperSlicer #1210: [FR] Minimum Travel Move Length

CCS86, 2021-05-31:
We already have a "minimum extrusion length" setting, which is functioning nicely in the latest release.

Could we also have a "minimum travel length" setting? This would become useful when the "avoid printed parts" setting is enabled. I find that with 2.3.56.1, I get travel segments just under 0.15mm long, even when minimum extrusion length is set to 0.3mm

qwewer0, 2021-06-01:
Could you explain the "minimum travel length" a bit more?

CCS86, 2021-06-01:
> Could you explain the "minimum travel length" a bit more?

It would just set a minimum threshold for the length of intermediate travel moves.  When you enable "avoid crossing perimeters", depending on the part, you will get curved travel moves. Like all curves, they are faceted. When you combine a commanded travel speed of say 200 mm/s, with travel move facets so small 0.15mm, you can end up with the demand for processing 1300+ moves per second. A printer can empty the buffer trying to accomplish this and stutter.

Defining a minimum travel segment length could alleviate this.

qwewer0, 2021-06-01:
> It would just set a minimum threshold for the length of intermediate travel moves. When you enable "avoid crossing perimeters", depending on the part, you will get curved travel moves. Like all [...]

I understand it now, thank you for the explanation.
And I think this would be a great addition to SS

CCS86, 2021-06-22:
Just an update on this issue. I ran into a problem on a pretty simple print, where suddenly the printer would make a loud bang, and suddenly layer shift. This was really weird, since there was no collision. I watched it happen a couple times, and it was repeatable. I tried upping stepper current, lowering travel accel & speed, but the issue kept happening.

Then I scrutinized where the failure occurred, and found it was on a print move, but just after a travel sequence made up of very tiny segments. I disabled "avoid crossing perimeters" and boom, problem gone. Something about running through so many move segments so fast (170mm/s), was causing the Duet / RRF motion planner to fail.

Hopefully this change can get coded in soon, since "avoid crossing perimeters" is a very useful feature. My duet board is likely on the sharp end of handling high segment rates, and still has a big issue with these tiny moves.

closed 2021-11-19 by supermerill