supermerill/SuperSlicer #1214: Feature request: Make the nozzle go into the outside of the corners to fill them up

pandaym3D, 2021-05-31:
We discussed this feature on Discord.

Intended to help avoid gaps in surfaces when printing with large(r) nozzles.

[image]

CCS86, 2021-06-01:
I'm curious if you have made some manual code to test the functionality of it.

I do a lot of CNC milling, where the absolute sharpest inside-corner you can make, is the radius of the cutter. This seems like the same principal to me. A nozzle orifice is circular, so the sharpest outside corner (radius) you can make, should be roughly [extrusion width] / 2. You would achieve that by taking a squared off path through the corner. Because of print head acceleration, you inevitably reduce speed through the corner, and getting a blip of over-extrusion bulge in the instant. 

I would tend to think that driving the nozzle further out from the corner would give you a shape like this:

[image]

supermerill, 2021-06-01:
yes, it's an experiment to see if movinig the nozzle there without extruding will move the extrusion by friction i the empty area.
It's not for the external perimeter, but more for inner ones, and preferably with 'external perimeter first'