supermerill/SuperSlicer #1215: Feature request: Alternative perimeter distribution

pandaym3D, 2021-05-31:
**Is your feature request related to a problem? Please describe.**
I would like to be able to control how the toolpaths prioritizes outer perimeters vs. holes, to ultimately be able to print stronger parts.

**Describe the solution you'd like**
Advanced perimeter number settings. I imagine being able to set:
- Outer perimeter count
- Hole perimeter total count
- Hole perimeter count (forced full circle)

Image below shows how I image it would slice a given part. In this example I would have set that all holes need one full circle perimeter, and a total of two perimeters. The rest goes to external perimeters. That way the perimeters stay outside the holes for the most of the part.
[image]

**Describe how it would work**
Here seen side by side with the current algo.
Instead of making perimeters curl behind holes, it should force them in front.
[image]

**Describe alternatives you've considered**
I tried to achieve this by using modifiers, but I was not able to.

**Additional context**
Image below show a picture of a part failure, that I believe can be avoided by being able to control perimeter distribution:
[image]

CCS86, 2021-06-01:
I like this idea.

My latest print:

[image]

supermerill, 2021-06-01:
What that could be
* print the external perimeter like now (outer & holes at the same time)
* print outer perimeters up to the outer number
* go back to the first hole perimeter and enlarge it a bit and clip it with the outer perimeters to get chunks of loops.
* get the small empty areas and fill them with gapfill.