supermerill/SuperSlicer #1216: Improvement: Option to have first layer fully filled to improve bed adhesion for detailed prints

jpedrick, 2021-06-01:
**Is your feature request related to a problem? Please describe.**
Sometimes for prints that have a very detailed first layer I get poor adhesion for small details. I can improve this by setting "brim inside holes" so that the entire first layer is covered inside the perimeters. However, this is a bit problematic as well, since the "inside holes" brims may also be small. The hotend head lays down some filament, but as it pulls away the tiny "inside holes" brim for a tiny hole comes along with it as it's more strongly connected to the hotend than the heat bed.

An example that's a bit tricky to print: https://www.thingiverse.com/thing:714484

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I would like the option to fill the entire first layer filled within the exterior perimeter with the "bottom infill pattern."

**Describe how it would work**
From a user interaction perspective it could work like:
1. If the first layer will be entirely filled with a brim, use the bottom infill pattern to fill the entire bottom layer of the print.
2. An option next to "brim inside holes" "simplify bottom layer by removing holes," which would remove holes for just the first layer.

**Describe alternatives you've considered**
It's difficult to think of alternatives as the problem is very physical. Printing the first layer very slowly may help, increasing first layer flow rate, bringing the nozzle closer to the print bed w/ z-offset, etc. Each has downsides I think in comparison to simplifying the first layer.

martinbudden, 2021-06-22:
@jpedrick , I don't quite understand this. Do you mean the entire first layer would be filled in, so you would then have to drill out the holes manually?

neophyl, 2021-06-22:
Just right click the object and add part.  Select either cube or cylinder depending on your part shape.  Then resize the added part to cover the area and make it 0.2mm thick.  Then use the drop to bed icon/control.  Your part now has a 1 layer thick bottom that will all be sliced as a solid area.

martinbudden, 2021-06-22:
@supermerill , I have an alternative suggestion.

As @jpedrick states, small details on on the first layer can have poor adhesion. They are fragile until they are joined up with the infill and can become unstuck for a number of reasons, including:
1. They have a small area and so little adhesion to the bed
2. Since they are small, they cool quickly and so can become unstuck
3. They can be dragged off the bed as the hotend moves away from printing them
4. If they are slightly proud, then can be knocked off the bed by a travel move

One way of dealing with this would be to connect the interior perimeters with the exterior perimeters. Consider this first layer:
[image]
The four small holes are fragile until the infill is laid down. However if they were connected to the exterior perimeter, like so:
[image]
They would be much more resilient. In particular
1. Their area is effectively much more
2. Even if they become slightly unstuck, they are held in place and can recover when the infill comes along
3. There is no travel move away from them, so they won't be dragged off the bed

I suggest an option called something like "Continuous external perimeter" to allow this (of course this may vary according to the topology of the object being printed, there may be more than one continuous external perimeter).

(Note that my illustration has a slight gap where between the two perimeter lines joining the circles to the exterior, that is just because it was an easy way to do the illustration, the actual implementation should not have such a gap.)

martinbudden, 2021-06-22:
I've just seen the "One-loop perimeters" option. This is very similar, except that the "long inside travel (from external to holes)" **is** extruded, so the infill has to take account of this.

supermerill, 2021-06-22:
why not.

jpedrick, 2021-06-27:
> @jpedrick , I don't quite understand this. Do you mean the entire first layer would be filled in, so you would then have to drill out the holes manually?
> 
> I've done this on occasion, by designing the part that way in OpenSCAD. It works, but drilling out the holes leaves rather ragged edges, since the holes have no perimeters.

@martinbudden I usually print the first layer fairly thin at around 0.08mm or less. I find that having the first layer thin improves bed adhesion since it's flexible enough to stretch and bend while cooling. In addition having the nozzle so close to the bed creates some pressure forcing the PLA into the bed a little. I then just use an exacto knife to gently remove the brims and the edges are razor sharp. So, I understand with a thicker first layer this wouldn't make as much sense.

What I'd prefer here is just a simplified first layer with whatever the bottom layer infill pattern is covering the entire bottom surface after the initial external perimeter is sketched out. Subsequent layers would have the hole perimeters, so no drilling required to remove the paper thin first layer from the holes and no ragged edges.

I like your idea above as well as setting the layer 1 travel speed #1339 . But I still think for the I posted at the top it still wouldn't print all that nicely.