supermerill/SuperSlicer #1217: Strange Solid infill

chripink, 2021-06-01:
Hi, 

I installed the last version of SuperSlicer 2.3.56 W64 and I have an issue with the Solid Infill. 

Solid Infill is generated near my perimeters as in the picture. 
Is it a bug or a feature that I have to desactivate (but I don't find where...)
[image]

The used profile is the one for the Sidewinder X1

Here the STL used is a benchy

Thank you,

Christophe

supermerill, 2021-06-01:
It's often an artefact of the number of bottom layers.
there already many issues about that.
Only way currently is to add a modifier to reduce the number of bottom layers on layers 4-end

chripink, 2021-06-01:
Ok :)

Thank you for your answer 👍

closed 2021-06-04 by supermerill