supermerill/SuperSlicer #1218: [FR] Plater improvement

uocnb, 2021-06-01:
**Is your feature request related to a problem? Please describe.**
There are no information or settings for Plater grid line spacing, all we can do is guessing how large it is, and it is not precision for manual arrangement.

**Describe the solution you'd like**
It would be great if we have UI settings for Plater with following properties:
1. Grid line spacing with major and minor size like S3D. The default would be 25 mm for major and 5 mm for minor.
2. Plater ruler with unit and numbers along X and Y Plater axis.
3. Display axis labels for XYZ.

**Additional context**
S3D Plater for reference, the bold line is major and lighter one is minor:
[image]

Visualization settings:
[image]

neophyl, 2021-06-01:
This is accomplished by using a custom bed image with whatever marking you want on it.

uocnb, 2021-06-01:
> This is accomplished by using a custom bed image with whatever marking you want on it.

Yes it is doable with custom bed image, but I think it is better to have this drew by the application itself. Easier with scale options and more precise than image.

supermerill, 2021-06-01:
that way?
[image]

uocnb, 2021-06-01:
@supermerill 👍
Much better compared with S3D, it is easier to me to distinguish from major line if the minor line is darker. WDYT?

supermerill, 2021-06-02:
> @supermerill 👍
> Much better compared with S3D, it is easier to me to distinguish from major line if the minor line is darker. WDYT?

currently, I set :
* 5mm: width 0.5, color 0.9f, 0.9f, 0.9f, 0.6f
* 10mm: width 1.5, color 0.9f, 0.9f, 0.9f, 0.6f
* 50mm: width 3, color 0.9f, 0.9f, 0.9f, 0.6f

do you want to test some variants?, I'll post pictures of them.

uocnb, 2021-06-02:
Thank @supermerill for quick update ⏩. I think the bed grid should not pop out too much from the background to not disturb the way we view the model above it. I tried to draw some samples with `pygame`, color is using `RGBA` with range `0-255`. It is not support float value for the line width hence I used integer instead (there might be workaround or other method, but I did this quick and dirty 😛). You can try with following sets to have better look.

Line width are same:
- 5 mm: width 1
- 10 mm: width 2
- 50 mm: width 3

Color set 1:
- Background: 60, 60, 60, 255
- 5 mm: 150, 150, 150, 80
- 10 mm: 120, 120, 120, 80
- 50 mm: 100, 100, 100, 50
[image]

Color set 2:
- Background: 60, 60, 60, 255
- 5 mm: 80, 80, 80, 30
- 10 mm: 100, 100, 100, 50
- 50 mm: 120, 120, 120, 20
[image]

Color set 3:
- Background: 153, 153, 153, 255
- 5 mm: 140, 140, 140, 30
- 10 mm: 130, 130, 130, 50
- 50 mm: 120, 120, 120, 255
[image]

Other thing is we can try with 2 type of line instead of 3. I found that it is a bit hard to distinguish the grid with 3. It would be great to have this as an option, so we can switch back and forth.

Sirage48, 2021-06-05:
I would prefer the Major Axes align with the origin - important for printers with the origin in the centre of the bed.

Thanks

supermerill, 2021-06-06:
oh, right

closed 2021-11-07 by supermerill