supermerill/SuperSlicer #1219: [FR] Sidebar quick settings

uocnb, 2021-06-01:
**Is your feature request related to a problem? Please describe.**
There are only `Supports`, `Infill` and `Brim` options are available on the sidebar. To change other options we need to go to each tab and search for it which take time and not convenience.

**Describe the solution you'd like**
More options for sidebar. Configurable is much better. Options should be from list of most used / change for each print like:
1. Layer height
2. Speed
3. Infill
4. Support (already exists)

**Describe alternatives you've considered**
Could be defined as print profiles like Cura:
1. Fast
2. Draft
3. Normal
4. Fine

**Additional context**
Cura settings for reference:
[image]

[image]

macdarren, 2021-06-02:
I second this, I find I also often need to change perimeter and top/bottom count and infill pattern.  I don't want too many quick access overrides, but the ones that significantly affect quality, strength, adhesion seem logical as it allows a single well tuned profile to be used while still creating a way to make a draft/quick print then later use the same profile set to make a final print.  Current I have what feels like too many profiles with settings scattered in three separate (print/filament/printer) files that must be used as a set.  I would not mind having the profiles and overrides used added to the comments at top of GCode but maybe that is a separate feature request.

martinbudden, 2021-06-22:
I third this.

Slic3r has a "Shortcuts" setting on the "Print settings" page, (under "Notes") this brings up a page of checkboxes, one for each setting - if you check the box then setting appears on the sidebar.

PrusaSlicer dropped this, no idea why, it was a really good feature.

I **really** miss this. Would it be possible to port it over from Slic3r?

neophyl, 2021-06-22:
Prusa Slicer was forked before that was added I think.  So it never had it.

DaveBoyce1, 2021-11-17:
Adding my vote. Absolutely would love to see this implemented. There are common 'go to' items that I'm alway changing, the list being mentioned already, plus ironing. Ended up here because I was searching how to do it remembering the ability from Slic3r days.

supermerill, 2021-11-17:
I looked a bit. The current implementation in PrusaSlicer (and so in SuSi) is heavily hardcoded.

supermerill, 2022-08-08:
Update: in latest version, you can now change your quick setting layout & options:
https://github.com/supermerill/SuperSlicer/issues/3020
If you only want to add an already existing setting without any simplification, take exemple over the infill ratio:
`setting:label$Infill:width$8:sidetext$  :fill_density`
`setting` is to start defining a setting
`label$Infill` is to change the displayed label in the sick setting bar
`width$8`is to reduce the widge size
`idetext$  ` is to remove any suffix.
`fill_density` is the name of the setting, as shown in the setting tooltip.

closed 2024-02-11 by supermerill