supermerill/SuperSlicer #1220: 2.3.56.1 - wxWidgets Debug Alert in Config Manager

Naugrimohtar, 2021-06-01:
**Describe the bug**
Running Super Slicer (2.3.56.1) using 'local settings, i.e. superslicer_local-settings.bat', freshly unzipped, using configuration manager. Select my Prusa Printers, then when choosing filament, when I clicked 'PLA', a pop up occurs with a debug warning. See screenshot for details. Repeatible.

**To Reproduce**
See description

**>> Project File <<**
Free unzip and stand-alone run, no configuration yet

**Expected behavior**
Not so pop up a debug window ;)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 2.3.56.1

**Additional context**
None

supermerill, 2021-06-01:
can reproduce in 2.3.56.1 but not in my dev build... I'll test the nightly when built.

jfestrada, 2021-06-06:
I can confirm that the bug still in the last release with the Windows x64 version, even with the [last nightly](https://github.com/supermerill/SuperSlicer/actions/runs/909628047). 

**Desktop (please complete the following information):**
* OS: Windows 10
* Version 2.3.56.1

**Additional context**
I have tested it with the PRUSA and VERTEX profiles

The Linux AppImage version is working fine.

closed 2021-07-10 by supermerill