supermerill/SuperSlicer #1222: [FR] XY Deceleration in Machine limits

qwewer0, 2021-06-01:
**Description**
If there would be a deceleration setting for X and Y axis in Printer Settings/Machine limits, that could make the time estimation to be a bit more precise, if a different value is used in the printer (Klipper), and if used for printing too, than it might reduce ringing/echo a bit.

supermerill, 2021-06-01:
why deceleration different than acceleration?

qwewer0, 2021-06-01:
> why deceleration different than acceleration?
> For the time estimation, it's marlin-based and I don't have time to tune it this year (and the next).

@supermerill
If we take a cube as an example, a relatively high acceleration/deceleration value, would cause more ringing on the following side after the corner, than with a lower value, but for acceleration a higher value is more forgiving for quality.

The main point is, that a lover deceleration value (half?) compared to the acceleration would result a cleaner print surface compared to the same high acc/dec values, or if we would keep the deceleration as default and only increase the acceleration, than the print time would shorten a bit, while the quality would mostly stay the same.

At least this is my idea...

supermerill, 2021-06-02:
I checked but I can't find a firmware that has a setting to control deceleration. And so I can't separate acceleration from deceleration.

qwewer0, 2021-06-02:
> I checked but I can't find a firmware that has a setting to control deceleration. And so I can't separate acceleration from deceleration.

In Marlin there isn't an option for that, yet.

But here is a [link](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#printer) for Klipper `max_accel_to_decel`.

supermerill, 2021-06-02:
> In Marlin there isn't an option for that, yet.
>
> But here is a [link](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#printer) for Klipper `max_accel_to_decel`.

that's not a decel command, but a kind of max_speed.

qwewer0, 2021-06-03:
> that looks like a kind of max_speed, not a decel command.

I will close this for now.

closed 2021-06-03 by qwewer0