supermerill/SuperSlicer #1223: Adding new filament features to the Filament Overrides

TheAndr0id, 2021-06-01:
**Describe the solution you'd like**
Some of the new settings and features would work well as filament overrides (Filament Settings/Filament Overrides) so they could be applied specifically for given filaments. 

Options/features that come to mind:
Print Settings/Width & Flow/Flow ratios (bridges/above the bridges/top fill/first layer)
Printer Settings/Retraction/Lift Z Enforcement

**Describe how it would work**
Exactly as the other filament overrides work with lift-z and retraction settings.

**Describe alternatives you've considered**
Right now I manually change the values when working with specific filaments.  It works, but when I forget it usually requires reprinting the part.  When you switch resolutions the warning of unsaved settings requires flipping through all the settings to figure out what's changed.

Sineos, 2021-06-02:
Elephant foot compensation is also a filament characteristic IMHO. 
Different material types have different z-offsets and react differently to squishing

supermerill, 2021-06-02:
Currently, filament overrides can only work for the extruder setting, as they are linked.
Big work.

supermerill, 2022-08-08:
#668