supermerill/SuperSlicer #1224: Slicing a "full plate" results in strange perimeters

weasel0x00, 2021-06-02:
**Describe the bug**
When Slicing a plate full of a bunch of different STLS / Objects, the perimters become erratic and inconsistent.  Reducing the number of objects (even by hiding / excluding them in the UI) resolves the issue

**To Reproduce**
Steps to reproduce the behavior:
Create a Plate with a lot of items, slice it and inspect the preview closely after slicing. I have attached images and a .3mf file if that helps.

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
This exists at least in the MacOS and linux versions of 2.3.56.0 and 2.3.56.1

**Additional context**
Add any other context about the problem here.

weasel0x00, 2021-06-02:
[image]
[image]
[image]

supermerill, 2021-06-02:
Also on prusa:
[image]

i checked the gcode: it's good

[image]

so it's a problem in the gcode viewer. Maybe too much things to draw?