supermerill/SuperSlicer #1226: Mac version switch to preview screen after entering a value in printer/filament settings

hagan66, 2021-06-02:
When background processing is on, the Mac version switches to preview screen after changing any value in the settings tabs. 

Version 2.3.56.1

neophyl, 2021-06-02:
It does that on all operating systems.  Its a natural side effect of having background processing on.  When you change something it re-slices.  As its sliced it takes you to the preview.  

You have 2 options , you can either turn background processing OFF (which is my preference as I only want it to slice when I tell it to) or there is a new option under Configuration>Preferences>Gui called 'Switch from 3d view to Preview when sliced" that is ticked by default.  If you untick it, then you wont get taken to the preview automatically at all.

hagan66, 2021-06-02:
Ok! Great, thanks! BTW Superslicers filament change temp control has made my MMU prints go without any interventions, from being a mess. Im so happy for this. :-)

closed 2021-06-02 by hagan66