supermerill/SuperSlicer #1227: Apple Silicon Support

dianlight, 2021-06-02:
**Is your feature request related to a problem? Please describe.**

Is not a problem, is an opportunity. ;)
PrusaSlicer support Apple Silicon native from version 2.3.1 [Changelog] (https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1) and the speed improvement is impressive.

**Describe the solution you'd like**

Release also a native compiled version for AARM64 architecture.

**Describe how it would work**
N/A

**Describe alternatives you've considered**
N/A

**Additional context**

If you need a beta tester for the AARM64 version I'm available

supermerill, 2021-06-02:
Planned for 2.3.57

supermerill, 2021-06-26:
I've merged the prusaslicer 2.3.1, so it should be able to compile on Apple Silicon.
Could you try? You can try the `BuildMacOS.sh` script, maybe it will compile on the first try, with a lot of luck.
If not, you'll have to tell me what to modify (or share your own script).

dianlight, 2021-06-26:
The first attempt failed! 👎 

I try the source released with v 2.3.56.6 but don't compiles...
Sound like a boost issue... maybe with an update...

```
[ 27%] Built target libslic3r_cgal
Consolidate compiler generated dependencies of target libslic3r
[ 28%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Arrange.cpp.o
In file included from /Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libslic3r/Arrange.cpp:21:
In file included from /Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/deps/build/destdir/usr/local/include/boost/multiprecision/integer.hpp:9:
In file included from /Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/deps/build/destdir/usr/local/include/boost/multiprecision/cpp_int.hpp:1979:
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/deps/build/destdir/usr/local/include/boost/multiprecision/cpp_int/limits.hpp:27:36: error: ambiguous conversion for functional-style cast from 'number<boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::unsigned_magnitude, boost::multiprecision::unchecked, void>, et_off>' to 'result_type' (aka 'number<boost::multiprecision::cpp_int_backend<128U, 128U, (cpp_integer_type)1U, (cpp_int_check_type)0U, void>, (boost::multiprecision::expression_template_option)0U>')
   static const result_type val = -result_type(~ui_type(0));
                                   ^~~~~~~~~~~~~~~~~~~~~~~
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/deps/build/destdir/usr/local/include/boost/multiprecision/cpp_int/limits.hpp:129:22: note: in instantiation of function template specialization 'std::detail::get_min<128, 128, boost::multiprecision::signed_magnitude, boost::multiprecision::unchecked, void, boost::multiprecision::et_off>' requested here
      return detail::get_min<MinBits, MaxBits, SignType, Checked, Allocator, ExpressionTemplates>(boost::multiprecision::backends::is_fixed_precision<backend_type>(), boost::multiprecision::is_signed_number<backend_type>());
                     ^
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/deps/build/destdir/usr/local/include/boost/integer/common_factor_rt.hpp:447:44: note: in instantiation of member function 'std::__1::numeric_limits<boost::multiprecision::number<boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::signed_magnitude, boost::multiprecision::unchecked, void>, boost::multiprecision::et_off>>::min' requested here
    if(a == (std::numeric_limits<Integer>::min)())
                                           ^
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libslic3r/Arrange.cpp:473:12: note: in instantiation of function template specialization 'libnest2d::minAreaBoundingBox<ClipperLib::Polygon, long long, boost::rational<boost::multiprecision::number<boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::signed_magnitude, boost::multiprecision::unchecked, void>, boost::multiprecision::et_off>>>' requested here
    return minAreaBoundingBox<S, TCompute<S>, boost::rational<LargeInt>>(sh)
           ^
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libslic3r/Arrange.cpp:516:26: note: in instantiation of function template specialization 'Slic3r::arrangement::min_area_boundingbox_rotation<ClipperLib::Polygon>' requested here
            itm.rotation(min_area_boundingbox_rotation(itm.rawShape()));
                         ^
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libslic3r/Arrange.cpp:643:5: note: in instantiation of function template specialization 'Slic3r::arrangement::_arrange<libnest2d::_Box<ClipperLib::IntPoint>>' requested here
    _arrange(items, fixeditems, to_nestbin(bed), params, pri, cfn);
    ^
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/deps/build/destdir/usr/local/include/boost/multiprecision/number.hpp:120:25: note: candidate constructor [with Other = boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::unsigned_magnitude, boost::multiprecision::unchecked, void>, ET = boost::multiprecision::et_off]
   BOOST_MP_FORCEINLINE number(const number<Other, ET>& val,
                        ^
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/deps/build/destdir/usr/local/include/boost/multiprecision/number.hpp:126:13: note: candidate constructor [with Other = boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::unsigned_magnitude, boost::multiprecision::unchecked, void>, ET = boost::multiprecision::et_off]
   explicit number(const number<Other, ET>& val, typename boost::enable_if_c<
            ^
In file included from /Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libslic3r/Arrange.cpp:8:
In file included from /Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libnest2d/include/libnest2d/placers/nfpplacer.hpp:18:
In file included from /Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libnest2d/include/libnest2d/placers/placer_boilerplate.hpp:4:
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libnest2d/include/libnest2d/nester.hpp:839:47: warning: lambda capture 'this' is not used [-Wunused-lambda-capture]
        if(infl > 0) std::for_each(from, to, [this, infl](Item& item) {
                                              ^
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libslic3r/Arrange.cpp:391:15: note: in instantiation of function template specialization 'libnest2d::_Nester<libnest2d::placers::_NofitPolyPlacer<ClipperLib::Polygon, libnest2d::_Box<ClipperLib::IntPoint>>, libnest2d::selections::_FirstFitSelection<Clip [... cut ...]
/Users/ltarantino/Downloads/SuperSlicer-2.3.56.6/src/libslic3r/Arrange.cpp:643:5: note: in instantiation of function template specialization 'Slic3r::arrangement::_arrange<ClipperLib::Polygon>' requested here
    _arrange(items, fixeditems, to_nestbin(bed), params, pri, cfn);
    ^
3 warnings and 1 error generated.
make[3]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/Arrange.cpp.o] Error 1
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make[1]: *** [src/CMakeFiles/Slic3r.dir/rule] Error 2
make: *** [Slic3r] Error 2

```

dianlight, 2021-06-26:
I made some research:
https://www.boost.org/doc/libs/1_76_0/libs/multiprecision/doc/html/boost_multiprecision/intro.html

You use the 1.70.0 version, in my mac the brew install the 1.76.0

I try change to 1.76.0 
boost_1_76_0.tar.gz | 7bd7ddceec1a1dfdcbdb3e609b60d01739c38390a5f956385a12f3122049f0ca
-- | --

Same result.
[Build.Slic3r.log](https://github.com/supermerill/SuperSlicer/files/6719668/Build.Slic3r.log)

mattapperson, 2021-06-26:
@supermerill confirming I too have seen the same issue as @dianlight on the master as of 5 min ago

claytono, 2021-11-14:
I've been able to build Superslicer locally on Monterey for Apple Silicon.  It seems like now the limitation would be building it with GitHub Actions, since Actions doesn't support apple silicon.  It should be possible to cross compile on an x86_64 MacOs worker for Apple Silicon.  I've poked at this a bit and made some progress, but still have some things to work out.

supermerill, 2021-11-14:
If you can direct me how to modify the BuildMacos.sh building script to cross-compile, I will be very happy.

claytono, 2021-11-15:
@supermerill I'll work on a PR when I get a bit further.  Right now I have some dumb things hard coded for testing purposes.  I've managed to get all the deps cross compiling, but ran into a make issue when building SS itself that I didn't have a chance to look into over the weekend.

claytono, 2021-11-18:
I've managed to get x86_64 builds on Apple Silicon working.  I did this first, since builds on M1 are so much faster than my x86 build host.  I'm working on getting Apple Silicon builds working on x86_64 now, which is tricker, since you can't run any of the binaries you produce there.    I've opened a [draft PR](https://github.com/supermerill/SuperSlicer/pull/1919) if anyone is interested.