supermerill/SuperSlicer #1232: Debug check failed when switching printer profiles

kaibsora, 2021-06-03:
**Describe the bug**
A clear and concise description of what the bug is.
when switching profiles, you are presented with a debug check has failed window.

**To Reproduce**
Steps to reproduce the behavior:

Upon unzipping application on windows, open the app
choose 2 printers and click finish
try to switch between printer a and printer b
debug check appears
vice versa when going from printer b to printer a

**Expected behavior**
no debug check?

**Screenshots**
screenshot attached
[image]
**Desktop (please complete the following information):**
 - OS:windows
 - Version 2.3.56.1

**Additional context**
you can click continue and it goes away. but it comes back whenever you switch.
[image]

supermerill, 2021-06-03:
can reproduce with 2.3.56.1 but not with the nigthly, should be fixed

closed 2021-06-13 by supermerill

kaibsora, 2021-06-16:
Thank you so much for the awesome work you do! I know you dont hear that enough!