supermerill/SuperSlicer #1233: Multiple Extruders - Layer height cant be greater than nozzle diameter

bung69, 2021-06-03:
### Version
2.3.56.1

### Operating system type + version
win10

### Behavior
Layer height cant be greater than nozzle diameter Error seems to only compare layer height against the first extruder nozzle diameter. rather than the extruder in use for the print.   

Error is triggered with a layer height above 0.8mm  when all extrusion types are set to extruder 3 with 1.4mm Nozzle  and extruder 1 has a nozzle diameter of 0.8mm

[Layer height error.zip](https://github.com/supermerill/SuperSlicer/files/6589357/Layer.height.error.zip)
[image]
[image]
[image]
[image]
[image]

supermerill, 2021-06-03:
can reproduce with 2.3.56.1 but not with the nigthly, should be fixed

bung69, 2021-06-03:
oops might of attached wrong file in the OP.  

[Layer height error.zip](https://github.com/supermerill/SuperSlicer/files/6589625/Layer.height.error.zip)

Just checked on the nightly,  empty error box appears.  and invalid data button,  however slice now button appears after toggling a setting but still doesn't work.

It also looks like there are other bugs in the nightly, such as brim ears not working and the Gcode and sliced Previews being very slow / Chugging.

supermerill, 2021-06-03:
> brim ears not working

i just fixed them, do you have a project for that bug?

bung69, 2021-06-08:
Project attached. 

Nozzle size / max layer Hight appears to work correctly with multiple extruders now Thanks, 
Gcode and sliced previews interaction both work at a decent speed now too. 

Brim Ears still dosnt work 
Support is on top of  a raft even with interface layers and raft set to 0,  I forget if this is expected / desired or not but it overlaps the brim.  i usually use the work around of "solid first layer" 

[Brim Ears and Support Raft Error.zip](https://github.com/supermerill/SuperSlicer/files/6614135/Brim.Ears.and.Support.Raft.Error.zip)

supermerill, 2021-06-08:
I see, the problem is that the brim ears are not generated around the support like the normal one.

bung69, 2021-06-09:
[image]

[image]

[image]

I get No Brim Ears at all,  on the part or the support.

closed 2021-06-09 by bung69