supermerill/SuperSlicer #1234: Supporting dense layer anchor at sparse infill lines

supermerill, 2021-06-03:
In v2.2.53, this still seems suboptimal...   Is there any reason we can't have a mode like automatic, but instead of expanding the area to a perimeter line, it simply expands to *any* line on the layer below (including an infill line).

That way, the area will never be expanded beyond the infill line spacing.

For some infill patterns, there might be certain angles where it is a *long* distance to the next infill line in the worst case.   In that case, a different direction for the dense layer support should be chosen.

The algorithm *seems* simple on the surface, but I guess implementation is harder than it looks...

_Originally posted by @Hello1024 in https://github.com/supermerill/SuperSlicer/issues/507#issuecomment-770128516_

supermerill, 2021-08-21:
just a note: infill are printed in parallel, so you don't have the "previous layer infill lines".

FT999, 2021-11-23:
Hello im new here :D 

So pleas forgive me if i do something wrong. 
For me would be perfect solution to this problem if the slicer coud great a denser infill on steep internal overhangs like it does if i set the setting "only infill where needed" to true. The rest of the infill could be much less dense.